darktable-org/rawspeed #253: Investigate DNG FillOrder=2 support

LebedevRI, 2020-11-23:
I think original @LebedevRI interpretation of specs that 24-bit data should always be big-endian is correct, because nowhere it's mentioned that 24-bit case should follow TIFF byte order. But I'm not arguing in reverting that commit, since it's about compatibility with other software. But I think FillOrder is indeed what matters here.

As to your request about files that aren't decoded correctly - I have DNG files generated by our custom software at work (can't release it publicly, at least for now) with FillOrder = 2. Motivation for this format is that most industrial cameras use little-endian packed format (see GenICam Pixel Format Naming Convention (PFNC), e.g. Mono10p format) and it's not always possible to transform the data on-the-fly. Right now these files aren't rendered correctly in darktable (as expected, since they don't conform to DNG specification). Will you be interested in implementing support for FillOrder tag in DNG? I don't think it can break support for normal DNG files (ignoring this 24 bit special case). I can of course provide sample files, both monochrome and bayer. If you're interested, I will create a feature request.

_Originally posted by @parafin in https://github.com/darktable-org/rawspeed/issues/252#issuecomment-732238409_

LebedevRI, 2020-11-23:
I have no idea if that can be supported or not, but are those images supported elsewhere?
I'm mainly worried about what we have just had in #252, that the data and metadata mismatch
and we "suddenly" misdecode things.

LibRaw, 2020-11-23:
> are those images supported elsewhere?

Quote from Adobe:
```
	if (fFillOrder != 1)
		{
		
		#if qDNGValidate

		ReportError ("Unsupported FillOrder",
					 LookupParentCode (parentCode));
					 
		#endif
					 
		return false;
		
		}
```

parafin, 2020-11-23:
No, I do't think they are supported by any software (I haven't seen one at least). But it also means that all other existing DNG files either don't have FillOrder (meaning default value of 1) or have it explicitly set to 1. So adding support for FillOrder=2 shouldn't affect any other (valid) DNG files.

Though indeed 24 bit stuff discussed in #252 affects this somewhat, but it can remain as-is IMHO. FillOrder=2 generally makes sense for little endian TIFF/DNG files, where the fix you committed for #252 already behaves like FillOrder=2 (at least in my interpretation of the standard).

parafin, 2020-11-23:
Bit order isn't "inverted" for any FillOrder, it's only about how they are split and packed. I've already drawn a diagram how they are written in #252. I'm not 100% sure it's correct, that's why I'm asking about how everybody else interpret the standard here.
I can upload a sample DNG and the same one converted to unpacked DNG format. I will share a link a little bit later.

LibRaw, 2020-11-23:
> Bit order isn't "inverted" for any FillOrder,

libtiff authors may disagree with this statement:
 #1: https://gitlab.com/libtiff/libtiff/-/blob/master/libtiff/tif_read.c#L283-286
 #2: https://gitlab.com/libtiff/libtiff/-/blob/master/libtiff/tif_swab.c#L284-301 (also see bit inversion table above)

LebedevRI, 2020-11-23:
> I can upload a sample DNG and the same one converted to unpacked DNG format. I will share a link a little bit later.

Please do. We won't make progress otherwise.

parafin, 2020-11-23:
> > I can upload a sample DNG and the same one converted to unpacked DNG format. I will share a link a little bit later.
> 
> Please do. We won't make progress otherwise.

http://paraf.in/junk/FillOrder2.dng - packed 12bit bayer DNG with FillOrder=2
http://paraf.in/junk/FillOrder2-unpacked.dng - same file unpacked to 16bits, reference of what was encoded in the previous file

Again, I'm not sure that format of the first file matches what FillOrder=2 actually means, and indeed @LibRaw is correct that libtiff wants to invert bits in each byte for this case, but I'm not convinced that's correct behaviour. Original TIFF specification only talks about 1-bit files, where indeed it's equivalent to bit reversal inside the byte.

LebedevRI, 2020-11-23:
Yeah no, like i expected, that's not *actually* FillOrder=2, you just messed up data endianness, 
beacuse rawspeed happily decodes it with the following change:
```
diff --git a/src/librawspeed/decompressors/AbstractDngDecompressor.cpp b/src/librawspeed/decompressors/AbstractDngDecompressor.cpp
index 9ca0b2b3..5ee3f6fe 100644
--- a/src/librawspeed/decompressors/AbstractDngDecompressor.cpp
+++ b/src/librawspeed/decompressors/AbstractDngDecompressor.cpp
@@ -55,16 +55,16 @@ template <> void AbstractDngDecompressor::decompressThread<1>() const noexcept {
 
     // DNG spec says that if not 8/16/32 bit/sample, always use big endian.
     // It's not very obvious, but that does not appear to apply to FP.
-    switch (mBps) {
-    case 8:
-    case 16:
-    case 32:
-      break;
-    default:
-      if (mRaw->getDataType() == TYPE_USHORT16)
-        big_endian = true;
-      break;
-    }
+    //     switch (mBps) {
+    //     case 8:
+    //     case 16:
+    //     case 32:
+    //       break;
+    //     default:
+    //       if (mRaw->getDataType() == TYPE_USHORT16)
+    //         big_endian = true;
+    //       break;
+    //     }
 
     try {
       const uint32_t inputPixelBits = mRaw->getCpp() * mBps;

```

LebedevRI, 2020-11-23:
I.e. to fix this you need to fix the code that generated the original DNG to
1. not write errneous FillLevel=2 :)
2. Follow DNG spec in the sense that since the data is not 8/16/32 bit, it should be stored in big-endian format.

parafin, 2020-11-23:
OK, using big-endian for data format isn't possible here, so I'm trying to understand if it's possible to fit into TIFF specification.
If we forget about DNG spec for a moment and that it says that FillOrder has to be 1, does FillOrder=2 change byte order?

LebedevRI, 2020-11-23:
Simplest solution: just pad the data up to the 8/16/32 bits.

LebedevRI, 2020-11-23:
@parafin, so pseudocode, but you want something like this:
https://godbolt.org/z/rj9xaj
I really don't understand why it wouldn't work for you?
Little-endian packing should be much more complicated.
I fear you are overthinking it. If you want, i'm sure you know my email.

parafin, 2020-11-23:
Yeah, that's obvious and not an option either. I think basically I'm out of luck, that's just not correct TIFF image data format. I'll close this ticket.

closed 2020-11-23 by parafin