darktable-org/rawspeed #261: Compiling in Visual Studio?

TomArrow, 2021-02-01:
I'm trying to use this in a project I have in Visual Studio, but as a C++ beginner I'm at the end of my wits for now.

The header files I managed to make work by doing this in a newly created rawspeedconfig.h:
```cpp
#ifdef _MSC_VER
#define __attribute__((x)) 
#define __attribute__(x) 
#define __builtin_unreachable(x)
//#define assert(x)
#define ASAN_REGION_IS_POISONED(x) 0
#define __PRETTY_FUNCTION__ __FUNCSIG__
#endif
```
I also copied a bunch of the code from config.h.in to my rawspeedconfig.h. After that the only remaining issue after compiling was unresolved symbols (all the actual function bodies etc).

But when I include all the other sources, it will just spit out a billion syntax errors and die. Most if it seems to be due to some gcc related stuff. I wanted to use this code as a submodule in git so I don't want to just go rogue and replace all the GCC stuff (and I'm not sure if I should either, since I would likely end up breaking something). 

Is there any way at all to compile this code without gcc?

I was thinking of compiling with gcc and using the static library inside Visual Studio but I then read that this is impossible, so I'm back to zero.

LebedevRI, 2021-02-01:
Only LLVM/Clang and GCC compilers are supported.
Microsoft's compiler is not supported, and i'm not sure what hacks/changes would be needed to make it work.

I would recommend trying to use clang-cl, see https://clang.llvm.org/docs/MSVCCompatibility.html

TomArrow, 2021-02-02:
Thanks for your answer.

I tried opening the folder in Visual Studio (2019 btw) as a CMake project, set the config to clang and I got this error:
STUDIO/2019/COMMUNITY/VC/Tools/Llvm/bin/clang-cl.exe has no C++14 support.

Do I need to take a more manual approach, so to speak?

I also got some error about a config.json not being able to be opened initially.

Then later I tried it again by just creating a static library project and adding the files. That made the C++14 error go away, but instead I got an error about that processor cores function in some CMakeLists.txt. Can you tell me how and where I must implement this? Is this some kind of "Cmake function" or whatever? Sorry if that sounds dumb. Or do I just create a normal C++ function like that?

LebedevRI, 2021-02-02:
Honestly, i have no idea. I have never used Visual Studio, and i have never used clang-cl.
I can guess that even with clang-cl, as per https://clang.llvm.org/docs/UsersManual.html#id9,
dash-style flags aren't avaliable, and forward-slash-style flags must be used.
I.e. https://github.com/darktable-org/rawspeed/blob/62f607357648ebce956f57b31b37e688afc86d0d/cmake/compiler-flags.cmake#L12 should be checking for `/std:14`, when the compiler is clang-cl.

TomArrow, 2021-02-02:
Going back to the original idea of compiling with VS normally (since clang doesn't seem all that well implemented in VS, not giving meaningful errors for example), would you consider doing a change or accepting a pull request to the code where the GCC specific stuff is turned into a macro that is defined conditionally based on the compiler? So for example, in the code itself, __attribute__((const)) would become ATTRIBUTE_CONST and then in some config file or something, it would be like this:
```cpp
#define ATTRIBUTE_CONST __attribute__((const))
```
and then I could just do my own define in Visual studio that goes:
```cpp
#define ATTRIBUTE_CONST 
```
which would in effect just turn it into empty space.

Because the problem I have is that the ones with double brackets like __attribute__((const)) can't get replaced via macro, since the double bracket is interpreted by Visual Studio as a syntax error. 

Of course I could just do my own fork and leave it like that there, but then I that would create long-term problems with maintaining the change etc., whereas implementing this directly into the main codebase would provide a tiny step towards being more generally compatible without sacrificing the original functionality this stuff was meant to provide. At least that's the way I see it from my limited experience. Let me know what you think.

LebedevRI, 2021-02-02:
I would be open/sympathetic to supporting clang-cl.
I'm not sure i'm interested in supporting VS compiler.

TomArrow, 2021-02-02:
Well, it wouldn't necessarily mean you have to support it explicitly, it would just get rid of some major hurdles that make it impossible. But alright, I'll respect your decision.

I'll try and look into the clang thing if I find the time and motivation. Thanks.

TomArrow, 2021-02-04:
So when I try to properly use the CMAKE gui and I specify the platform toolset as "ClangCL", I get a bit further than usual, but I end up with the same problem with the compiler flag test. 
Here's the exact errors I get:
```
CMake Warning at cmake/build-type.cmake:3 (message):
  CMAKE_BUILD_TYPE is not defined!
Call Stack (most recent call first):
  CMakeLists.txt:185 (include)

WARNING: Defaulting to CMAKE_BUILD_TYPE=Sanitize. Use ccmake to set a proper value.
```
This error doesn't stop the build, it's just marked in red, so I thought I'd mention it.

Here's the fatal one:
```
CMake Error at cmake/compiler-flags.cmake:14 (message):
  The compiler C:/Program Files (x86)/Microsoft Visual
  Studio/2019/Community/VC/Tools/Llvm/x64/bin/clang-cl.exe has no C++14
  support.  Please use a different C++ compiler.
Call Stack (most recent call first):
  CMakeLists.txt:208 (include)
```
I tried to comment out this line:
```
message(FATAL_ERROR "The compiler ${CMAKE_CXX_COMPILER} has no C++14 support. Please use a different C++ compiler.")
```
and it got a bit further but ended up failing at:
```
CMake Error at cmake/Modules/GoogleTest.cmake:12 (message):
  CMake step for googletest failed: 1
Call Stack (most recent call first):
  cmake/src-dependencies.cmake:8 (include)
  CMakeLists.txt:232 (include)
```

I strongly suspect I shouldn't be editing that line out, but at least it helped me find another error.

TomArrow, 2021-02-04:
Not sure if it helps in any way but here's the -help output of the clang-cl.exe:
```
OVERVIEW: clang LLVM compiler

USAGE: clang-cl.exe [options] file...

CL.EXE COMPATIBILITY OPTIONS:
  /?                      Display available options
  /arch:<value>           Set architecture for code generation
  /Brepro-                Write current time into COFF output (default)
  /Brepro                 Do not write current time into COFF output (breaks link.exe /incremental)
  /clang:<arg>            Pass <arg> to the clang driver
  /C                      Do not discard comments when preprocessing
  /c                      Compile only
  /d1PP                   Retain macro definitions in /E mode
  /d1reportAllClassLayout Dump record layout information
  /diagnostics:caret      Enable caret and column diagnostics (default)
  /diagnostics:classic    Disable column and caret diagnostics
  /diagnostics:column     Disable caret diagnostics but keep column info
  /D <macro[=value]>      Define macro
  /EH<value>              Set exception handling model
  /EP                     Disable linemarker output and preprocess to stdout
  /execution-charset:<value>
                          Set runtime encoding, supports only UTF-8
  /E                      Preprocess to stdout
  /fallback               Fall back to cl.exe if clang-cl fails to compile
  /FA                     Output assembly code file during compilation
  /Fa<file or dir/>       Set assembly output file name (with /FA)
  /Fe<file or dir/>       Set output executable file name
  /FI <value>             Include file before parsing
  /Fi<file>               Set preprocess output file name (with /P)
  /Fo<file or dir/>       Set output object file (with /c)
  /fp:except-             
  /fp:except              
  /fp:fast                
  /fp:precise             
  /fp:strict              
  /Fp<file>               Set pch file name (with /Yc and /Yu)
  /GA                     Assume thread-local variables are defined in the executable
  /Gd                     Set __cdecl as a default calling convention
  /GF-                    Disable string pooling
  /GF                     Enable string pooling (default)
  /GR-                    Do not emit RTTI data
  /Gregcall               Set __regcall as a default calling convention
  /GR                     Emit RTTI data (default)
  /Gr                     Set __fastcall as a default calling convention
  /GS-                    Disable buffer security check
  /GS                     Enable buffer security check (default)
  /Gs                     Use stack probes (default)
  /Gs<value>              Set stack probe size (default 4096)
  /guard:<value>          Enable Control Flow Guard with /guard:cf, or only the table with /guard:cf,nochecks
  /Gv                     Set __vectorcall as a default calling convention
  /Gw-                    Do not put each data item in its own section (default)
  /Gw                     Put each data item in its own section
  /GX-                    Deprecated (like not passing /EH)
  /GX                     Deprecated; use /EHsc
  /Gy-                    Do not put each function in its own section (default)
  /Gy                     Put each function in its own section
  /Gz                     Set __stdcall as a default calling convention
  /help                   Display available options
  /imsvc <dir>            Add <dir> to system include search path, as if in %INCLUDE%
  /I <dir>                Add directory to include search path
  /J                      Make char type unsigned
  /LDd                    Create debug DLL
  /LD                     Create DLL
  /link <options>         Forward options to the linker
  /MDd                    Use DLL debug run-time
  /MD                     Use DLL run-time
  /MTd                    Use static debug run-time
  /MT                     Use static run-time
[... 225 lines cut ...]
                          Minimum time granularity (in microseconds) traced by time profiler
  -ftime-trace            Turn on time profiler. Generates JSON file based on output filename.
  -ftrivial-auto-var-init=<value>
                          Initialize trivial automatic stack variables: uninitialized (default) | pattern
  -fvirtual-function-elimination
                          Enables dead virtual function elimination optimization. Requires -flto=full
  -fwhole-program-vtables Enables whole-program vtable optimization. Requires -flto
  -gcodeview-ghash        Emit type record hashes in a .debug$H section
  -gcodeview              Generate CodeView debug information
  -gdwarf                 Generate source-level debug information with the default dwarf version
  -gline-directives-only  Emit debug line info directives only
  -gline-tables-only      Emit debug line number tables only
  -gno-inline-line-tables Don't emit inline line tables
  -miamcu                 Use Intel MCU ABI
  -mllvm <value>          Additional arguments to forward to LLVM's option processing
  -nobuiltininc           Disable builtin #include directories
  -print-supported-cpus   Print supported cpu models for the given target (if target is not specified, it will print the supported cpus for the default target)
  -Qunused-arguments      Don't emit warning for unused driver arguments
  -R<remark>              Enable the specified remark
  --target=<value>        Generate code for the given target
  --version               Print version information
  -v                      Show commands to run and use verbose output
  -W<warning>             Enable the specified warning
  -Xclang <arg>           Pass <arg> to the clang compiler
```

Ben-Voris, 2021-02-04:
I know nothing about Microsoft C++ or clang, and only a very little about cmake, but I *think* cmake is trying to test if the compiler you've told it to use supports the C++14 standard. Presumably Darktable has code that uses C++14 features and has told cmake that C++14 is required. For an example, see [How can I add a minimum compiler version requisite?](https://stackoverflow.com/questions/14933172/how-can-i-add-a-minimum-compiler-version-requisite). 

These two pages might be helpful: [Microsoft C++ language conformance table](https://docs.microsoft.com/en-us/cpp/overview/visual-cpp-language-conformance?view=msvc-160) and this [Enabling C++14 in clang in Visual Studio](https://stackoverflow.com/questions/29951318/enabling-c14-in-clang-in-visual-studio/29951851). If I understand the latter correctly and if I assume that it is accurate, the compiler should enable the latest support it provides. But, it might be worth playing with `/std:<value>`.

It could also be that clang-cl.exe does not report version information in a way that this cmake file understands.

LebedevRI, 2021-02-12:
> I tried to comment out this line:

You'll just need to add fallbacks to all such places in cmake to also try `/windows`-style flags as last-resort.

> Not sure if it helps in any way but here's the -help output of the clang-cl.exe:

Not to me. Someone who cares about that platform and has direct access to that compiler should be doing such porting.
I have already suggested how that particular place should be fixed:

> ```
>   /std:<value>            Set C++ version (c++14,c++17,c++latest)
> ```

TomArrow, 2021-02-28:
So, I've created my own fork and resorted, at least temporarily, to commenting out the part that sets the flag parameter. The reason is this:
```# yes, need to keep both the CMAKE_CXX_FLAGS and CMAKE_CXX_STANDARD.
# with just the CMAKE_CXX_STANDARD, try_compile() breaks:
#   https://gitlab.kitware.com/cmake/cmake/issues/16456
# with just the CMAKE_CXX_FLAGS, 'bundled' pugixml breaks tests
#   https://github.com/darktable-org/rawspeed/issues/112#issuecomment-321517003
```
The issue referenced when leaving out the CMAKE_CXX_FLAGS appears to be resolved on the CMake tracker so perhaps this is no longer necessary. But if there's evidence to the contrary, I'll be happy to give it another try.

In any case, what currently stops me from progressing is the googletest issue. 
```
# Download and unpack googletest at configure time
set(GOOGLETEST_PREFIX "${RAWSPEED_BINARY_DIR}/src/external/googletest")
configure_file(${RAWSPEED_SOURCE_DIR}/cmake/Modules/GoogleTest.cmake.in ${GOOGLETEST_PREFIX}/CMakeLists.txt @ONLY)

execute_process(COMMAND ${CMAKE_COMMAND} -G "${CMAKE_GENERATOR}"
  -DALLOW_DOWNLOADING_GOOGLETEST=${ALLOW_DOWNLOADING_GOOGLETEST} -DGOOGLETEST_PATH:PATH=${GOOGLETEST_PATH} .
  RESULT_VARIABLE result
  WORKING_DIRECTORY ${GOOGLETEST_PREFIX}
)

if(result)
  message(FATAL_ERROR "CMake step for googletest failed: ${result}")
endif()
```
The error thrown by cmake is: ```CMake step for googletest failed: 1```

I tried looking into the folder in the binary output folder src/external/googletest but found no source code or anything, just some Cmake stuff. There's also a log file and some other stuff in there but nothing that provides any clue to what went wrong.

Any ideas?

TomArrow, 2021-02-28:
One thing I think might be the issue is commands like this:
```
set(GOOGLETEST_PATH "/usr/src/googletest" CACHE PATH
                    "Path to the googletest root tree. Should contain googletest and googlemock subdirs. And CMakeLists.txt in root, and in both of these subdirs")

set(PUGIXML_PATH "/usr/src/pugixml" CACHE PATH "Path to the pugixml root tree.")

set(ZLIB_PATH "/usr/src/zlib" CACHE PATH "Path to the zlib root tree.")

set(GOOGLEBENCHMARK_PATH "/usr/src/googlebenchmark" CACHE PATH
                    "Path to the googlebenchmark root tree.")
```

Obviously on Windows there can't be a /usr/src/...

My spontaneous solution to this would be to create a subfolder "ext" under "src" and add git submodules with the required libraries. Thoughts?

LebedevRI, 2021-02-28:
> So, I've created my own fork and resorted, at least temporarily, to commenting out the part that sets the flag parameter. The reason is this:
> 
> ```
> # with just the CMAKE_CXX_STANDARD, try_compile() breaks:
> #   https://gitlab.kitware.com/cmake/cmake/issues/16456
> # with just the CMAKE_CXX_FLAGS, 'bundled' pugixml breaks tests
> #   https://github.com/darktable-org/rawspeed/issues/112#issuecomment-321517003
> ```
> 
> The issue referenced when leaving out the CMAKE_CXX_FLAGS appears to be resolved on the CMake tracker so perhaps this is no longer necessary. But if there's evidence to the contrary, I'll be happy to give it another try.

... which was part of cmake-3.8, while rawspeed requires 3.10.
So yeah, i'd take a PR cleaning that up.

> <...googletest...>

Works just fine in windows-based CI.
As messages should have said, you'll need to either disable building tests,
or point it at the location of sources, or allow downloading them.
You probably want latter. I don't want to pull it via submodule.

TomArrow, 2021-02-28:
Ah, interesting. Okay so my change would then basically be to remove this part:
```message(STATUS "Checking for -std=c++14 support")
CHECK_CXX_COMPILER_FLAG("-std=c++14" COMPILER_SUPPORTS_CXX14)
if(NOT COMPILER_SUPPORTS_CXX14)
  message(FATAL_ERROR "The compiler ${CMAKE_CXX_COMPILER} has no C++14 support. Please use a different C++ compiler.")
else()
  set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++14")
  message(STATUS "Checking for -std=c++14 support - works")
endif()
```
And replace the message about needing both with a short note that that is no longer necessary. Does that make sense? I'm still very much beginning to understand how cmake works so wanna make sure I'm not breaking anything. So far I only commented that part out.

About googletest, thanks, I learned a new thing. Realized there's a checkbox I can set in the CMAKE GUI to activate the downloading and that worked. Curiously, it worked only for googletest and not for the other stuff like zlib, openmp, etc. When I put those checkboxes there, they just get ignored and after the next round of "config", they are deselected again. 

I also get a bunch of messages from googletest about how some kind of deprecation. Might need to update to a newer googletest version or something? But doesn't seem like a serious error so I'll let that one go.

In other news, deactivating the additional libraries at least made the config succeed and I was able to build the projects. Further, I was able to compile the rawspeed subproject but I had to manually remove the "-g0" and "-O3" flags from the compiler settings. Doing that for the single subproject is okay but for the full build of hundreds of projects that might be very obnoxious to do manually. I presume those are gcc-only flags? Clang CL said that -g0 is an unknown argument and that -O3 is an "unused argument", whatever that means.

In any case, I was able to compile a .lib and successfully load it into the linker in my other project. Well, not sure if successfully, but at least the compilation finished, which I'm gonna call a success.

LebedevRI, 2021-02-28:
> Ah, interesting. Okay so my change would then basically be to remove this part:
> 
> ```
> CHECK_CXX_COMPILER_FLAG("-std=c++14" COMPILER_SUPPORTS_CXX14)
> if(NOT COMPILER_SUPPORTS_CXX14)
>   message(FATAL_ERROR "The compiler ${CMAKE_CXX_COMPILER} has no C++14 support. Please use a different C++ compiler.")
> else()
>   set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++14")
>   message(STATUS "Checking for -std=c++14 support - works")
> endif()
> ```
> 
> And replace the message about needing both with a short note that that is no longer necessary. Does that make sense? I'm still very much beginning to understand how cmake works so wanna make sure I'm not breaking anything. So far I only commented that part out.

As per, https://gitlab.kitware.com/cmake/cmake/-/commit/f72ba42b7c643b1b217d1b8f6684cec3289e7201#9735efb2f01e32e5293f52150b5e52561787a4ce,
i think so.

> In other news, deactivating the additional libraries at least made the config succeed and I was able to build the projects. Further, I was able to compile the rawspeed subproject but I had to manually remove the "-g0" and "-O3" flags from the compiler settings.

See above, i have already answered that exact question before.

TomArrow, 2021-02-28:
Alright, I'll prepare a PR if/when I find the time. Haven't done one for a while.

Re. the flags, are you referring to your comment about windows-style flags? Bc all the other flags in the VS project are listed with "-" as well and seem to cause no issues, it's only those two he doesn't seem to know what to do with.

WindowsNT, 2021-12-15:
Any news on VS cmake? Or perhaps I can manually add the files to a new project?

LebedevRI, 2021-12-15:
There are no news because no one has volounteered to do the porting.
I would personally strongly suggest to just use clang-cl.
Note that if #325 happens, clang will explicitly become the only supported compiler.

WindowsNT, 2021-12-15:
99,9% of Visual Studio Windows developers use msvc.  The only feasible solution for them therefore would be to build some DLL with rawspeed and even so, this could cause major incompatibilies when two different compilers are used in the same project. 

Apart from that, this seems a great library and it's a pity I can't use it as of now, but will perhaps try to work on it later.

LebedevRI, 2021-12-15:
It may come off being blunt, but non-llvm based compilers are a

LebedevRI, 2021-12-15:
> 99,9% of Visual Studio Windows developers use msvc.

Good for them.