darktable-org/rawspeed #262: Streaming or Tile support

ryanstout, 2021-03-07:
First, rawspeed looks like a great library, thanks to all who worked hard on it. I've been reading through the source and I think I know the answer, but is it possible to access raw image data without full loading the resulting cfa channels into memory? (for example to load the image in tiles or by scanline or something). I'm combining multiple raw files at a time and need to only load part of them in ram at once. Any help would be appreciated. If its not possible, how difficult would it be to modify rawspeed (I could probably fund some development if its something thats not a huge change) Thanks!

LebedevRI, 2021-03-08:
Hi.

That is indeed not supported.
Which file/image sizes are we talking about?
What container, compression?

I suppose something like specifying interesting region-of-interest
that should only be decoded could be theoretically feasible,
but i'm not quite sure how big of a change that would be.

There's also a known chicken-and-the-egg problem,
how'd you know which area of an image you'll want to ask for,
if currently the only way to know the image size is to decode it?

ryanstout, 2021-03-08:
@LebedevRI thanks for the info. I would need .nef, .cr2, and .awr support ideally. For me it's really about being able to load only tiles of the image at a time. (or sets of scanlines would work also). Since I'm processing multiple raw photos at once with limited ram, the device doesn't have the ram to load everything at once. (Currently, we're doing a load on each photo, then writing out tiles to disk and then loading those tiles as we process, but thats a lot of extra disk IO)

So as far as asking for what area, it could be as simple as having it give me the first 256 scanlines, then the next 256, etc..  (Or whatever the block sizes are for DCT based codecs) Looking at the source I figured it would be a big change, figured it couldn't hurt to ask incase I was missing something.

I'm sure you're super busy, but if you have any interest in some paid dev work on this, message me, ryan at witharsenal.com and we can chat.

Thanks!

LebedevRI, 2021-03-08:
Hm, i was hoping you'd say DNG, and gigapixel image size :/