darktable-org/rawspeed #264: Phase One IQ150 sample picture

hrehfeld, 2021-03-22:
The Phase One IQ150 seems to be unsupported, here's a[ sample file](http://haukerehfeld.de/phaseone_iq150.zip).

I was told to ping @LebedevRI  :smile:

LebedevRI, 2021-03-22:
Hi.
The camera samples need be on https://raw.pixls.us/ so they do not get lost, and can be used by everyone who needs them.

hrehfeld, 2021-03-22:
Oh, sorry, I was looking for instructions, but the readme here didn't have anything.

Uploaded there, but it didn't give me an url or anything... Maker/Model were detected correctly though.

LebedevRI, 2021-03-22:
Doesn't that camera have other RAW mode[s]?
As per https://downloads.phaseone.com/645DFplus_IQ-Series_User_Guide_1.7-36a0.pdf
i think there should be both IIQ L, IIQ S.
if it does, it would be best to have both, with same image content.

laptop006, 2023-04-08:
I've just uploaded a hopefully full set of sample files for the IQ150 to raw.pixls.us.

Given some of the other Phase One support bugs I've uploaded a HDR set so there's some heavy over and underexposure references if that helps.

Phase One IQ1 50MP on a Phase One XF (both running latest firmware), Schneider 80mm f2.8
ISO 100, f/8.
No filters on the lens.

Two triple exposure HDR sets, one in IIQ-L, one IIQ-S
1/2500 (underexposed 3 stops)
1/320 
1/40 (overexposed 3 stops)