darktable-org/rawspeed #265: Support Quad Bayer Remosaic to bayer?

hanjtcv, 2021-04-01:
hi, Now dose the project support Quad Bayer Remosaic to bayer pattern? I couldn't find the information. Looking forward to your reply.
Function as follows.
[image]

LebedevRI, 2021-04-01:
There is no handling for Quad Bayer CFA whatsoever.
Which camera produces such raws?
Is there a sample on https://raw.pixls.us/?

hanjtcv, 2021-04-01:
sony imx 586 and samsung S5KHMX can output quad bayer pattern.

LebedevRI, 2021-04-01:
I'm sorry, i do not understand the question.

> There is no handling for Quad Bayer CFA whatsoever.

If you have more specific camera support question,
please ensure that there is an appropriate sample on https://raw.pixls.us/ and reopen this.

closed 2021-04-01 by LebedevRI

kmilos, 2021-04-02:
> sony imx 586 and samsung S5KHMX can output quad bayer pattern.

These are sensors, not cameras.

AFAIK, no camera (e.g. smartphone containing these sensors, or any 3rd party camera app running on a smartphone containing these sensors) provides raw Quad Bayer CFA output...

If you know otherwise, please provide samples.