darktable-org/rawspeed #283: Exif data from Silverfast DNG causes rawspeed to fail

0x10, 2021-07-09:
This is a follow up issue from the darktable issue (https://github.com/darktable-org/darktable/issues/9447).

tldr of 9447:
- Silverfast (filmscanner) may produce DNG files with additional exif data (IFD1 + PreviewTiff) which causes rawspeed to report the image as corrupt.
- sample images found [here](https://github.com/darktable-org/darktable/issues/9447#issuecomment-876953077)

First thought, as shared within the 9447, was that because of the IFD1 Block rawspeed may fail. So I tried to remove it.
I ran `exiftool -all= <file>` and magically rawspeed (accessed via darktable) could read the image and all went well. Hurray!
Now I tried to find out whats different within the files.

I made a diff of the `exiftool -v` output (see attached) and the IFD1 and PreviewTiff Blocks are still present, but what was removed was:
- IFD0:ApplicationNotes
- IFD0:ExifOffset

I tried removing those by hand. IFD0:ApplicationNotes could be removed, but didn't change anything. The image was then still reported as corrupt.
IFD0:ExifOffset could not be removed by hand with exiftool (it says "not writeable"). Does anyone has a tip how to remove it?

But as ExifOffset remains, I guess its whats causing the error and I guess its the content of the MakerNoteUnknownBinary.

Any Ideas how to check that?

Here are the `exiftool -v` outputs and diffs.
[exif_meta_vs_no_meta_diff.log](https://github.com/darktable-org/rawspeed/files/6791072/exif_meta_vs_no_meta_diff.log)
[exif_w_meta.log](https://github.com/darktable-org/rawspeed/files/6791073/exif_w_meta.log)
[exif_wo_app_note.log](https://github.com/darktable-org/rawspeed/files/6791074/exif_wo_app_note.log)
[exif_wo_meta.log](https://github.com/darktable-org/rawspeed/files/6791075/exif_wo_meta.log)

kmilos, 2021-08-11:
Don't think it's metadata per se, more the file structure itself.

rawspeed fails [here](https://github.com/darktable-org/rawspeed/blob/4f03e0d42b1842f18b30a4e225e3bd39cbe00939/src/librawspeed/tiff/TiffIFD.cpp#L120) when parsing the additional IFD1 which does not exist in the lower resolution variant, but I don't think the file is really "corrupt" (exiftools manages to follow and parse the interleaved IFDs and SubIFDs ok), but the layout is a bit unusual...

kmilos, 2021-08-11:
Ah, there is indeed an erroneous overlap detected. Somehow there is an extra (non-existant) IFD after the Exif subIFD, although the Exif subIFD is properly terminated...

kmilos, 2021-08-11:
> I guess its the content of the MakerNoteUnknownBinary

@0x10 It was a good hunch after all, but unfortunately I also wasn't able to remove just this tag using exiftool...

LibRaw, 2021-08-12:
BTW, I've looked into makernotes (0x927c) parsing code. I see completely broken logic here:
 - makernotes tag data is analysed for known prefixes (e.g. OLYMP or Nikon)
 - if nothing known found  then entire tag data is passed to TIFF IFD parser
 
Then,  in TIFF IFD parser:
  - numEntries are read as 16bit int. For non-TIFF (binary) IFD this reads some random unsigned number
  - IFDFullSize is calculated as 6 + 12 * numEntries (this is some random number based on previous read)
  - IFDFullSize is **not** compared to makernotes tag size (makernotes size is not passed here, so no value to compare), so some random-size file section will be parsed.
  - Buffer(offset, IFDFullSize) is inserted into ifds set: this buffer is _random size_ so next IFD may overlap with it, resulting in (false) file refusal
   - and this Buffer get parsed via parseIFDEntry calls for numEntries items.
   - Again, we're parsing some binary (non TIFF-like) data, so wrong tag data types (this is just some binary bytes) may result into TIFFParserException and so in (correct) file refusal.

The easiest way to improve it is to add 'LSI1\0' as known binary-makernotes prefix and do not parse such sections.

This does not negate the need to correct the entire makernotes parsing logic to not fail on non-tiff makernotes:

- IFD parser should compare 0x927c tag size with IFDFullSize calculated and stop makernotes parsing (but do not refuse the file) if calculated size is larger than tag size (opposite case is OK).
- IFD overlap check should be fixed for that case: passed tag size (+offset) should not overlap with other ifds, not calculated size.
- TIFF tag parser should not stop/throw exception if makernotes section being parsed: makernotes are, by definition, vendor specific. It is possible that non-vendor parser does not know some vendor specific. In such case parser should skip unknowns, not raise fatal error.

Dixi

LibRaw, 2021-08-12:
Actually we're discussing raw data decoder, not general-purpose metadata parser (like exiftool or exiv2).
I'm not sure at all if it makes sense to parse unknown makernotes blocks: known blocks contains known data (e.g. black/white level) that may be used for raw processing. But what is supposed to be done with unknown data from unknown blocks?

Of course, changing the logic will require additional checks: from such a vendor we parse only such a block (with fixed known prefix list) and these tables will have to be maintained, which is a lot of work.
But this is better than rejecting the correct dng

0x10, 2021-09-01:
Thanks for both of your summaries (sorry for the late reply, holiday-season...). I'm though a bit confused about the status of the issue now. Do you have a possible solution and its only pending on implementation? Or is this a "won't fix" because its a vendor thingy?
Is there something I can do, to support?

kmilos, 2021-09-01:
> Or is this a "won't fix" because its a vendor thingy?

No, the file is perfectly valid.

> I'm though a bit confused about the status of the issue now. Do you have a possible solution and its only pending on implementation?

Yep, pending until someone comes up with a proposed fix for rawspeed.

0x10, 2021-09-01:
Okay thank you!