darktable-org/rawspeed #284: need canon eos m6 mark ii support

jheidemann, 2021-07-25:
I see cr3 support is on the canon_cr3 pull request (yea!).

That pull request has many cameras, but not the EOS M6 Mark II.  It seems like data/cameras.xml needs to be informed about this new model.  What is required to do so?

kmilos, 2021-07-26:
Duplicate of https://github.com/darktable-org/darktable/issues/3375

jheidemann, 2024-11-08:
Here is a patch adding support.  It was given to me by Daniel Vogelbacher (thank you!).
It works for me when I build darktable with it.

(the attachment is a patch, but has a .txt extension because gihub lies when they say they accept .patch attachments)

[darktable_3_8_1_canon_m6_mark_ii_241107.txt](https://github.com/user-attachments/files/17674265/darktable_3_8_1_canon_m6_mark_ii_241107.txt)