darktable-org/rawspeed #285: White level for Canon EOS Rebel T3

photopea, 2021-08-09:
I have this CR2 photo. [Your XML](https://github.com/darktable-org/rawspeed/blob/develop/data/cameras.xml) says, that the maximum white level is 15500. However, the "overexposed strip" (e.g. the pixel X=2200,Y=1100) has a value of 13584. 

Because of this, Darktable does not perform any correction of the overexposed area, and it is displayed as pink in Darktable.

Where did you get the value 15500 from? If it is wrong, could you fix it?

[IMG_1449.zip](https://github.com/darktable-org/rawspeed/files/6954947/IMG_1449.zip)

LebedevRI, 2021-08-09:
https://github.com/darktable-org/rawspeed/blob/1ad9c15546dfe207538c1f7aa3f7c73854b4ccf1/data/cameras.xml#L982  you mean?
Since there's no per-ISO levels, that's obviously broken (much like for the rest of canon cameras).
We could get one sample per ISO from that camera and hardcode/refine the values.
The real fix would be to read the black/white levels from makernotes, i think it's `0x02d4 SpecularWhiteLevel`
(though i'm not sure what `0x02d3 NormalWhiteLevel` and `0x02d5 LinearityUpperMargin` are for).

photopea, 2021-08-09:
Yes. So do you think you could fix it somehow? Have you tried to open my CR2 file in Darktable?

photopea, 2021-08-10:
According to EXIFTool, there are parameters

    0x02d3 Normal White Level              : 11767
    0x02d4 Specular White Level            : 12279

But none of values is close to the brightness value of 13584.

LebedevRI, 2024-03-18:
We now read them from makernotes.

closed 2024-03-18 by LebedevRI