darktable-org/rawspeed #303: #294 breaks compile on Ubuntu 18.04

wpferguson, 2021-09-08:
Ubuntu 18.04 still uses cmake 3.10.2.  18.04 still has 19 months unttil EOL. #294 changed the minimum to 3.13.

Editing the CMakeLists.txt file and changing the version back results in successful compilation.

LebedevRI, 2021-09-08:
Hi.

As per https://wiki.ubuntu.com/Releases, Ubuntu 18.04 is indeed LTS,
with "End of Standard Support" advertised as April 2023 (18 months from now).

However, the fact whether or not particular ubuntu distro is EOS/EOL
only matters to the ubuntu itself and the packages provided for that particular distro.
It does not mean that every future version of every project that is packaged in that release
must be buildable on that version of that particular distro.

Generally, i try to be very reasonable with required versions,
and try to, if there isn't a good reason not to, not require ones newer than
a. 3 years old
b. available in debian stable and latest ubuntu lts

In this case, all of the above is satisfied.

https://apt.kitware.com/ might be helpful.

wpferguson, 2021-09-08:
I understand.  Thanks for the pointer to https://apt.kitware.com/.  That worked.

closed 2021-09-08 by wpferguson

LebedevRI, 2021-09-08:
I'm sorry for the problems, i understand that such things are never fun.