darktable-org/rawspeed #327: Migrate darktable's adobe_coeff.c into cameras.xml

LebedevRI, 2021-12-09:
This has been brought up several times now, i think it should just happen.

LebedevRI, 2021-12-10:
I guess the general form should be
```
		<ColorMatrices>
			<ColorMatrix vendor="<???>" illuminant="21" rows="3" divident="10000">
				<ColorMatrixRow y="0" values="10000,10000,10000"/>
				<ColorMatrixRow y="1" values="10000,10000,10000"/>
				<ColorMatrixRow y="2" values="10000,10000,10000"/>
			</ColorMatrix>
		</ColorMatrices>
```
though perhaps without `vendor` for now.

kmilos, 2021-12-11:
Why skip vendor? Could leave to signify whether it came from Adobe, dcraw or custom measurement (could even include something for which chart was used, which regression method...)

Isn't borh "rows" and "y" redundant? One just parses the structure as is...

And it should be "divisor" or "denominator": https://en.m.wikipedia.org/wiki/Quotient

LebedevRI, 2021-12-11:
Missing reading part, but this is basically it: abf9114932b5cc743d3755027f381dab03658c9d.

closed 2021-12-14 by LebedevRI

kmilos, 2021-12-14:
Roman, the DNG ColorMatrix2 story is a bit more complicated than this: there is **zero guarantee** that the D65 calibration goes second, the DNG spec merely __recommends__ it.

> DNG provides for one, two, or three sets of color calibration tags, each set optimized for a different illuminant. If two or three sets of color calibration tags are included, then the raw converter should interpolate between the calibrations based on the white balance selected by the user.

> If exactly two calibrations are included, then it is recommended that one of the calibrations be for a low color temperature illuminant (e.g., Standard-A) and the second calibration illuminant be for a higher color temperature illuminant (e.g., D55 or D65).

Note that this subtle wording ("one of the") does not mandate the order of illuminants (i.e. not "the first"). There are numerous DNG samples where they are stored the other way around.

So you have to check both illuminant 2 & 1 for D65 basically. There is also illuminant 3 now in [DNG 1.6](https://helpx.adobe.com/content/dam/help/en/photoshop/pdf/dng_spec_1_6_0_0.pdf) (introduced yesterday).

> If three calibrations are included, then it is recommended that the first two calibrations are as described in the preceding paragraph, and the third calibration be used to characterize the "as shot" illuminant: that is, the illuminant used to photograph the main image. This third calibration provides a color transform optimized for the capture conditions, while providing additional calibrations for two other illuminants in case the user wishes to change the white balance significantly during raw conversion.

Again, not the __recommended__. In fact, if present and really corresponding to the correctly estimated "as shot" conditions, this might be preferred over returning the D65 matrix.

LebedevRI, 2021-12-14:
Cool

kmilos, 2021-12-14:
So basically IMHO rawspeed should expose all matrices (and illuminants or convert those to CCT value) found in a DNG, and let the raw processor pick which one to use instead of preferring D65 (you can put that one first/default if you want).

LebedevRI, 2021-12-14:
That's what the commit message said, yes.

LebedevRI, 2021-12-14:
6f446a9b185cf6da334c54c37209c200020a704a says:
```
This is a(n incomplete) Minimal Viable Product.
<...>

Many more things (e.g. vendor/illuminant/denominator attributes)
could be added once needed, but this is the minimal thing that just
completely subsumes the entirety of darktable's `adobe_coeff.c`

<...>

This adds rudimentary support for reading color matrices out of DNG's,
but only if the illuminant is 21, and the srational has divisor of 10'000.
Both, especially the second one, restrictions are bugs,
but they should be good enough for now.
```

I'm not sure what is confusing about "incomplete" and "rudimentary".

kmilos, 2021-12-14:
Roger that

commit 6f446a9b (2021-12-14): Import color matrices from darktable's `adobe_coeff.c` (Fixes #327)

This is a(n incomplete) Minimal Viable Product.
For all intents and purposes, unless specified otherwise,
it's intended to be identical to the `ColorMatrix2` DNG tag.

Many more things (e.g. vendor/illuminant/denominator attributes)
could be added once needed, but this is the minimal thing that just
completely subsumes the entirety of darktable's `adobe_coeff.c`

The values are implicitly fixed point with denominator of 10'000,
this is done intentionally (floating point is just nasty),
and matches the contents (or rather, the [... cut]
commit 58975e0e (2021-12-14): Merge remote-tracking branch 'upstream/develop' into stable

* (!) Mark RPU sample lacking entries in cameras.xml as unsupported.
* Sony ZV-E10 camera support
* Sony ILCE-7M4 camera support
* Import color matrices from darktable's `adobe_coeff.c`
* Basic support for reading color matrices out of DNG's.

* upstream/develop:
  RafDecoder: extract applyCorrections() out of decodeMetaDataInternal()
  DngDecoder::decodeMetaDataInternal(): const-qualify loop variable
  Import color matrices from darktable's `adobe_coeff.c` (Fixes #327)
  rawspeed: Fujifilm GFX50S II support
  Add Sony ILCE-7M4 camer [... cut]