darktable-org/rawspeed #328: Add support for Sona A7 IV

cryptomilk, 2021-12-11:
Hi Roman,

I've just uploaded RAW files for the A7M4 to raw.pixls.us. Let me know if you need any additional pictures.

Thanks!

kmilos, 2021-12-11:
See also https://github.com/darktable-org/darktable/issues/10556

LebedevRI, 2021-12-11:
Feel free to submit patches.

kmilos, 2021-12-11:
Not supported by ADC yet...

kmilos, 2021-12-12:
Unsurprisingly, [the same "WTH Sony" gotchas are present as in the ILCE-1 case for the new lossless mode](https://github.com/darktable-org/darktable/issues/8523):

1. Both tile and strip-based storage tags are present in the full resolution IFD, when they should be mutually exclusive (tile only)
2. Image size is unnecessarily increased to integer multiple of tile size

kmilos, 2021-12-14:
FYI, the dngmeta.sh tool output for cameras.xml is this:

```xml
        <Camera make="SONY" model="ILCE-7M4">
                <ID make="Sony" model="ILCE-7M4">Sony ILCE-7M4</ID>
                <CFA width="2" height="2">
                        <Color x="0" y="0">RED</Color>
                        <Color x="1" y="0">GREEN</Color>
                        <Color x="0" y="1">GREEN</Color>
                        <Color x="1" y="1">BLUE</Color>
                </CFA>
                <Crop x="0" y="0" width="0" height="0"/>
                <Sensor black="512" white="16383"/>
        </Camera>
```

but I won't be submitting a patch as I don't know the exact methodology preferred here to determine the crop.

cryptomilk, 2021-12-14:
@kmilos Can you also paste the adobe coeff (color matrix)?

kmilos, 2021-12-14:
Sure, it was already merged above, but here goes:

`{ "Sony ILCE-7M4", { 7460,-2365,-588,-5687,13442,2474,-624,1156,6584 } },`

P.S. Don't use auto-close keywords in rawspeed, as it messes up Roman's [triage system](https://github.com/orgs/darktable-org/projects/1).

cryptomilk, 2021-12-14:
Ah, they are already in darktable master. Nice :-)

PR #329 (2021-12-14): Add Sony ILCE-7M4 camera support
Fixes #328
closed 2021-12-14 by LebedevRI