darktable-org/rawspeed #331: [Q] Upcoming requirements bump

LebedevRI, 2021-12-16:
As discussed a number of times over the mail, now that the darktable major release is almost out,
after it happens, i would like to proceed with bumping the versions of deps.
Regardless of the halide situation, which is going to throw a massive wrench at all this, i want to bump at least to:
* GCC 10
* LLVM 10
* apple xcode 12.0 (the first xcode to be based on LLVM 10) / minimal OSX deployment target 10.14

@parafin will this be a problem on osx's side?

parafin, 2021-12-16:
fine by me (for macOS)

LebedevRI, 2021-12-16:
I must admit i find that surprising to hear.

LebedevRI, 2021-12-16:
@parafin just so i better understand the OSX/Xcode situation, what would be the highest versions that are still fine?
Is Xcode 12.5 (first one LLVM11 based) too much? 13.0 (first one LLVM12 based)?
Likewise for deployment target.

parafin, 2021-12-16:
Xcode 12.5 is too much, it requires macOS 11.0, which I can't install on my Mac. As for deployment target - I would prefer to keep it at 10.14, but if you have a good reason you can bump it up to 10.15, not higher.

LebedevRI, 2021-12-16:
I see, thank you. 12.0/10.14 seems fine, right now i don't have a reason for 10.15 yet.

closed 2021-12-24 by LebedevRI