darktable-org/rawspeed #340: sysctlbyname is not declared on OpenBSD while trying to query CPU L1d cache line size

xsgb, 2022-01-26:
Hello,

For the context, I'm working to get an openmp enabled build of darktable for OpenBSD.
I've managed to build and install a gcc-8 supporting openmp. It's correctly detected during the darktable's configuration phase.

However it turns that the method employed to get the CPU cache line size fails there.

Here's the log.

```
-- Trying to query CPU L1d cache line size
CMake Error at src/external/rawspeed/cmake/Modules/cpu-cache-line-size.cmake:15 (message):
  Failed to query CPU L1d cache line size:

  Change Dir: /usr/obj/ports/darktable-3.6.1/build-amd64/CMakeFiles/CMakeTmp

  

  Run Build Command(s):/usr/local/bin/ninja cmTC_3d436 && [1/2] Building CXX
  object CMakeFiles/cmTC_3d436.dir/cpu-cache-line-size.cpp.o

  FAILED: CMakeFiles/cmTC_3d436.dir/cpu-cache-line-size.cpp.o

  /usr/obj/ports/darktable-3.6.1/bin/c++ -O2 -pipe -msse3 -march=native
  -I/usr/local/include/Imath -Wall -Wformat -Wformat-security -Wshadow
  -Wtype-limits -Wvla -Wno-unknown-pragmas -Wno-error=varargs
  -Wno-format-truncation -w -std=c++17 -Wall -Wextra -Wcast-qual -Wextra
  -Wextra-semi -Wformat=2 -Wpointer-arith -Wvla -Wmissing-format-attribute
  -Wsuggest-attribute=format -Wno-unused-parameter -Wstack-usage=4096
  -Wframe-larger-than=4096 -Wlarger-than=32768 -fPIE -std=c++17 -o
  CMakeFiles/cmTC_3d436.dir/cpu-cache-line-size.cpp.o -c
  /usr/obj/ports/darktable-3.6.1/darktable-3.6.1/src/external/rawspeed/cmake/Modules/cpu-cache-line-size.cpp

  
  /usr/obj/ports/darktable-3.6.1/darktable-3.6.1/src/external/rawspeed/cmake/Modules/cpu-cache-line-size.cpp:
  In function 'int main()':

  
  /usr/obj/ports/darktable-3.6.1/darktable-3.6.1/src/external/rawspeed/cmake/Modules/cpu-cache-line-size.cpp:26:7:
  error: 'sysctlbyname' was not declared in this scope

     if (sysctlbyname("hw.cachelinesize", &val, &size, NULL, 0) != 0)
         ^~~~~~~~~~~~

  
  /usr/obj/ports/darktable-3.6.1/darktable-3.6.1/src/external/rawspeed/cmake/Modules/cpu-cache-line-size.cpp:26:7:
  note: suggested alternative: 'ctlname'

     if (sysctlbyname("hw.-- Trying to query CPU L1d cache line size
CMake Error at src/external/rawspeed/cmake/Modules/cpu-cache-line-size.cmake:15 (message):
  Failed to query CPU L1d cache line size:

  Change Dir: /usr/obj/ports/darktable-3.6.1/build-amd64/CMakeFiles/CMakeTmp

  

  Run Build Command(s):/usr/local/bin/ninja cmTC_3d436 && [1/2] Building CXX
  object CMakeFiles/cmTC_3d436.dir/cpu-cache-line-size.cpp.o

  FAILED: CMakeFiles/cmTC_3d436.dir/cpu-cache-line-size.cpp.o

  /usr/obj/ports/darktable-3.6.1/bin/c++ -O2 -pipe -msse3 -march=native
  -I/usr/local/include/Imath -Wall -Wformat -Wformat-security -Wshadow
  -Wtype-limits -Wvla -Wno-unknown-pragmas -Wno-error=varargs
  -Wno-format-truncation -w -std=c++17 -Wall -Wextra -Wcast-qual -Wextra
  -Wextra-semi -Wformat=2 -Wpointer-arith -Wvla -Wmissing-format-attribute
  -Wsuggest-attribute=format -Wno-unused-parameter -Wstack-usage=4096
  -Wframe-larger-than=4096 -Wlarger-than=32768 -fPIE -std=c++17 -o
  CMakeFiles/cmTC_3d436.dir/cpu-cache-line-size.cpp.o -c
  /usr/obj/ports/darktable-3.6.1/darktable-3.6.1/src/external/rawspeed/cmake/Modules/cpu-cache-line-size.cpp

  
  /usr/obj/ports/darktable-3.6.1/darktable-3.6.1/src/external/rawspeed/cmake/Modules/cpu-cache-line-size.cpp:
  In function 'int main()':

  
  /usr/obj/ports/darktable-3.6.1/darktable-3.6.1/src/external/rawspeed/cmake/Modules/cpu-cache-line-size.cpp:26:7:
  error: 'sysctlbyname' was not declared in this scope

     if (sysctlbyname("hw.cachelinesize", &val, &size, NULL, 0) != 0)
         ^~~~~~~~~~~~

  
  /usr/obj/ports/darktable-3.6.1/darktable-3.6.1/src/external/rawspeed/cmake/Modules/cpu-cache-line-size.cpp:26:7:
  note: suggested alternative: 'ctlname'

     if (sysctlbyname("hw.cachelinesize", &val, &size, NULL, 0) != 0)
         ^~~~~~~~~~~~
         ctlname

  ninja: build stopped: subcommand failed.
cachelinesize", &val, &size, NULL, 0) != 0)
         ^~~~~~~~~~~~
         ctlname

  ninja: build stopped: subcommand failed.

```

Link to the code: https://github.com/darktable-org/rawspeed/blob/88e2b92a204f2d404d156b0ba11a66604685d5a0/cmake/Modules/cpu-cache-line-size.cpp#L26

I've attached the complete log.

[log.txt](https://github.com/darktable-org/rawspeed/files/7946116/log.txt)

LebedevRI, 2022-01-27:
Patches welcomed :)

Quick googling suggests that https://man.openbsd.org/sysctl.2 needs to be used,
but then with which parameters should it be called?

xsgb, 2022-01-27:
Thanks for your reply!

`sysctl(2)` would be perfect except that this CPUID information is not get on OpenBSD currently.

To workaround I simply returned the l1 line size value I got from a machine cpuid call.

```
#include <stdio.h>
// This returns 64 on my AMD Ryzen 7 PRO 3700U
int main()
{
        uint32_t linesize;
        __asm__("cpuid"
                        : "=a" (linesize)
                        : "0" (5));
        printf("%d\n",linesize);
        return 0;
}
```

References:

https://en.wikipedia.org/wiki/CPUID#CPU-specific_information_outside_x86
https://github.com/gcc-mirror/gcc/blob/master/gcc/config/i386/cpuid.h
http://www.flounder.com/cpuid_explorer2.htm#CPUID(0x80000005)

LebedevRI, 2022-01-27:
> `sysctl(2)` would be perfect except that this CPUID information is not available on OpenBSD currently.
> There's no `_SC_LEVEL1_DCACHE_LINESIZE` as well as others 0x8000005/6 cache informations defined.
> For information, I did not see such definitions for other BSD's. Maybe I've not looked so deep, but at first glance, there's no `hw.cachelinesize` as well...

I'm not sure, but since we are talking about OpenBSD, let's talk about OpenBSD specifically.
I don't suppose they'd want to just provide the right interface in the first place?

xsgb, 2022-01-27:
That's right. Implementing the CPUID information config in the OS would be nice.
Let me reopen this when I have more informations.

This ticket is just a record for WIP towards OpenMP and OpenCL on those platforms if that's effectively related.

Thanks you for your time. Have a nice day!

closed 2022-01-27 by xsgb