darktable-org/rawspeed #343: failure to process certain DNG files (Leica M246)

ubergeek801, 2022-02-09:
Darktable/Rawspeed handle about 95% of the DNG files from a Leica M Monochrom (Typ 246) just fine, but fail on some of them. The failing files can be opened in RawTherapee, so this doesn't seem to be a matter of data corruption.

Using the `rstest` utility produces some output:

```
L1000625.DNG failed: /home/jeremy/dev/git/rawspeed/src/librawspeed/decompressors/AbstractDngDecompressor.cpp:224: void rawspeed::AbstractDngDecompressor::decompress() const: Too many errors encountered. Giving up. First Error:
/home/jeremy/dev/git/rawspeed/src/librawspeed/io/Buffer.h:173: const uint8_t* rawspeed::Buffer::getData(rawspeed::Buffer::size_type, rawspeed::Buffer::size_type) const: Buffer overflow: image file may be truncated
Total decoding time: 0s
```

If there are other diagnostics I can run, I'd be happy to do so. I can of course provide an offending DNG file as well (they tend to be on the order of 20 MB).

LebedevRI, 2022-02-09:
Could you please contribute one such image to https://raw.pixls.us/ ? (please do write a comment there about the problem)

ubergeek801, 2022-02-09:
I uploaded an image and it seemed to be successful, but I don't see it appearing in the repository. I'll check on it later and, if necessary, try again...

ubergeek801, 2022-02-11:
Strangely, the file is still not appearing, and if I try to upload it again, I am told that the file already exists. Hopefully I can get this sorted out...

ValZapod, 2022-04-15:
The only present file is tiff... https://raw.pixls.us/data/Leica/LEICA%20M%20MONOCHROM%20(Typ%20246)/

ubergeek801, 2022-04-15:
pixls still refuses to accept my upload, with the message "The file either already exists, the CC0/modified checkboxes weren't checked, or you forgot to choose a file." The file should have the name `L1000625.DNG`.

I'll try reaching out to Pat David (pixls.us owner) and see if I can get a response.

ubergeek801, 2022-04-15:
For what it's worth, this issue occurred with, I believe, three different images the first time I used the camera (about 80 shots, if I recall). Given the upload issue, I wish I had kept the other two!

I've taken perhaps another 100 shots with this camera since then, and the issue hasn't re-occurred. So I couldn't entirely rule out some sort of data corruption, although that hypothesis doesn't provide a satisfying explanation of RawTherapee's ability to process the file...

ValZapod, 2022-04-16:
This looks like that verification failes that is done by libraw. So maybe you can just attach it here?

ZaquL, 2022-05-29:
looks like a dup: https://github.com/darktable-org/darktable/issues/11695#issuecomment-1120988605

LebedevRI, 2024-01-16:
Ok, well, better than nothing i suppose...

@ubergeek801 thank you for contributing the samples!