darktable-org/rawspeed #354: GFX 100S 16 bits white point

jas01, 2022-05-08:
When importing a 16 bits RAW uncompress file from the GFX100S into darktable, they are a problem with the white level. Every image are way too bright.

To fix this problem someone give me a solution and it's to change in the cameras.xml
```xml
<Sensor black="63" white="16383"/>
```
to
```xml
 <Sensor black="63" white="65535"/>
```
But when looking in the cameras.xml that would also affect 14bits file. 

So is they are any way to differentiate the GFX 100S 16 bits with the 14 bits ?, so darktable pickup the right white point

kmilos, 2022-05-08:
See also https://github.com/darktable-org/rawspeed/issues/228

dllu, 2022-09-11:
I can confirm that this fix is needed even after checking out the latest `develop` branch of `rawspeed` and compiling Darktable.

LebedevRI, 2023-02-05:
I believe this was fixed by #421.

closed 2023-02-05 by LebedevRI