darktable-org/rawspeed #358: Debug build fails on GCC8

aurelienpierre, 2022-05-11:
Output from building 154081cbcf9a9141aa3508bbbe88fe8a810fcb6e:

```
FAILED: lib/darktable/rawspeed/CMakeFiles/rawspeed.dir/src/librawspeed/decompressors/VC5Decompressor.cpp.o 
/usr/bin/g++-8 -DDEBUG -DGDK_DISABLE_DEPRECATED -DGTK_DISABLE_DEPRECATED -DGTK_DISABLE_SINGLE_INCLUDES -D_GLIBCXX_SANITIZE_VECTOR -D_XOPEN_SOURCE=700 -D__GDK_KEYSYMS_COMPAT_H__ -I/home/runner/work/R-Darktable/R-Darktable/build/lib/darktable/rawspeed -I/home/runner/work/R-Darktable/R-Darktable/src/src/external/rawspeed -I/home/runner/work/R-Darktable/R-Darktable/build/lib/darktable/rawspeed/src -I/home/runner/work/R-Darktable/R-Darktable/src/src/external/rawspeed/src/librawspeed -isystem /home/runner/work/R-Darktable/R-Darktable/src/src/external/rawspeed/src/external -Wall -Wformat -Wformat-security -Wshadow -Wtype-limits -Wvla -Wno-unknown-pragmas -Wno-error=varargs -Wno-format-truncation -w -std=c++17 -Wall -Wextra -Wcast-qual -Wextra -Wextra-semi -Wformat=2 -Wpointer-arith -Wvla -Wmissing-format-attribute -Wsuggest-attribute=format -Wno-unused-parameter -Wstack-usage=4096 -Wframe-larger-than=4096 -Wlarger-than=32768 -Werror -g -O0 -fPIC -fvisibility=hidden -fvisibility-inlines-hidden   -march=native -g3 -ggdb3 -fopenmp -std=c++17 -MD -MT lib/darktable/rawspeed/CMakeFiles/rawspeed.dir/src/librawspeed/decompressors/VC5Decompressor.cpp.o -MF lib/darktable/rawspeed/CMakeFiles/rawspeed.dir/src/librawspeed/decompressors/VC5Decompressor.cpp.o.d -o lib/darktable/rawspeed/CMakeFiles/rawspeed.dir/src/librawspeed/decompressors/VC5Decompressor.cpp.o -c /home/runner/work/R-Darktable/R-Darktable/src/src/external/rawspeed/src/librawspeed/decompressors/VC5Decompressor.cpp
In file included from /usr/include/c++/8/cassert:44,
                 from /home/runner/work/R-Darktable/R-Darktable/src/src/external/rawspeed/src/librawspeed/decompressors/VC5Decompressor.cpp:39:
/home/runner/work/R-Darktable/R-Darktable/src/src/external/rawspeed/src/librawspeed/decompressors/VC5Decompressor.cpp: In member function ‘void rawspeed::VC5Decompressor::Wavelet::ReconstructableBand::createLowpassReconstructionTask(const bool&)’:
/home/runner/work/R-Darktable/R-Darktable/src/src/external/rawspeed/src/librawspeed/decompressors/VC5Decompressor.cpp:297:7: error: ‘__PRETTY_FUNCTION__’ not specified in enclosing ‘task’
       assert(highlow.has_value() && lowlow.has_value() &&
       ^~~~~~
/home/runner/work/R-Darktable/R-Darktable/src/src/external/rawspeed/src/librawspeed/decompressors/VC5Decompressor.cpp:288:9: error: enclosing ‘task’
 #pragma omp task default(none)                                                 \
         ^~~
In file included from /usr/include/c++/8/cassert:44,
                 from /home/runner/work/R-Darktable/R-Darktable/src/src/external/rawspeed/src/librawspeed/decompressors/VC5Decompressor.cpp:39:
/home/runner/work/R-Darktable/R-Darktable/src/src/external/rawspeed/src/librawspeed/decompressors/VC5Decompressor.cpp: In member function ‘void rawspeed::VC5Decompressor::Wavelet::ReconstructableBand::createHighpassReconstructionTask(const bool&)’:
/home/runner/work/R-Darktable/R-Darktable/src/src/external/rawspeed/src/librawspeed/decompressors/VC5Decompressor.cpp:323:7: error: ‘__PRETTY_FUNCTION__’ not specified in enclosing ‘task’
       assert(highhigh.has_value() && lowhigh.has_value() &&
       ^~~~~~
/home/runner/work/R-Darktable/R-Darktable/src/src/external/rawspeed/src/librawspeed/decompressors/VC5Decompressor.cpp:314:9: error: enclosing ‘task’
 #pragma omp task default(none)                                                 \
         ^~~
In file included from /usr/include/c++/8/cassert:44,
                 from /home/runner/work/R-Darktable/R-Darktable/src/src/external/rawspeed/src/librawspeed/decompressors/VC5Decompressor.cpp:39:
/home/runner/work/R-Darktable/R-Darktable/src/src/external/rawspeed/src/librawspeed/decompressors/VC5Decompressor.cpp: In member function ‘void rawspeed::VC5Decompressor::Wavelet::ReconstructableBand::createLowHighPassCombiningTask(const bool&)’:
/home/runner/work/R-Darktable/R-Darktable/src/src/external/rawspeed/src/librawspeed/decompressors/VC5Decompressor.cpp:353:7: error: ‘__PRETTY_FUNCTION__’ not specified in enclosing ‘task’
       assert(lowpass.has_value() && highpass.has_value() &&
       ^~~~~~
/home/runner/work/R-Darktable/R-Darktable/src/src/external/rawspeed/src/librawspeed/decompressors/VC5Decompressor.cpp:344:9: error: enclosing ‘task’
 #pragma omp task default(none)                                                 \
         ^~~
In file included from /usr/include/c++/8/cassert:44,
                 from /home/runner/work/R-Darktable/R-Darktable/src/src/external/rawspeed/src/librawspeed/decompressors/VC5Decompressor.cpp:39:
/home/runner/work/R-Darktable/R-Darktable/src/src/external/rawspeed/src/librawspeed/decompressors/VC5Decompressor.cpp: In member function ‘virtual void rawspeed::VC5Decompressor::Wavelet::AbstractDecodeableBand::createDecodingTasks(rawspeed::ErrorLog&, bool&)’:
/home/runner/work/R-Darktable/R-Darktable/src/src/external/rawspeed/src/librawspeed/decompressors/VC5Decompressor.cpp:581:9: error: ‘__PRETTY_FUNCTION__’ not specified in enclosing ‘task’
         assert(!decodedData.has_value() && "Decoded this band already?");
         ^~~~~~
/home/runner/work/R-Darktable/R-Darktable/src/src/external/rawspeed/src/librawspeed/decompressors/VC5Decompressor.cpp:573:9: error: enclosing ‘task’
 #pragma omp task default(none) shared(decodedData, errLog, exceptionThrown)    \
         ^~~
[95/916] /usr/bin/g++-8 -DDEBUG -DGDK_DISABLE_DEPRECATED -DGTK_DISABLE_DEPRECATED -DGTK_DISABLE_SINGLE_INCLUDES -D_GLIBCXX_SANITIZE_VECTOR -D_XOPEN_SOURCE=700 -D__GDK_KEYSYMS_COMPAT_H__ -I/home/runner/work/R-Darktable/R-Darktable/build/lib/darktable/rawspeed -I/home/runner/work/R-Darktable/R-Darktable/src/src/external/rawspeed -I/home/runner/work/R-Darktable/R-Darktable/build/lib/darktable/rawspeed/src -I/home/runner/work/R-Darktable/R-Darktable/src/src/external/rawspeed/src/librawspeed -isystem /home/runner/work/R-Darktable/R-Darktable/src/src/external/rawspeed/src/external -Wall -Wformat -Wformat-security -Wshadow -Wtype-limits -Wvla -Wno-unknown-pragmas -Wno-error=varargs -Wno-format-truncation -w -std=c++17 -Wall -Wextra -Wcast-qual -Wextra -Wextra-semi -Wformat=2 -Wpointer-arith -Wvla -Wmissing-format-attribute -Wsuggest-attribute=format -Wno-unused-parameter -Wstack-usage=4096 -Wframe-larger-than=4096 -Wlarger-than=32768 -Werror -g -O0 -fPIC -fvisibility=hidden -fvisibility-inlines-hidden   -march=native -g3 -ggdb3 -fopenmp -std=c++17 -MD -MT lib/darktable/rawspeed/CMakeFiles/rawspeed.dir/src/librawspeed/decoders/AbstractTiffDecoder.cpp.o -MF lib/darktable/rawspeed/CMakeFiles/rawspeed.dir/src/librawspeed/decoders/AbstractTiffDecoder.cpp.o.d -o lib/darktable/rawspeed/CMakeFiles/rawspeed.dir/src/librawspeed/decoders/AbstractTiffDecoder.cpp.o -c /home/runner/work/R-Darktable/R-Darktable/src/src/external/rawspeed/src/librawspeed/decoders/AbstractTiffDecoder.cpp
ninja: build stopped: subcommand failed.
+ cmake --build /home/runner/work/R-Darktable/R-Darktable/build -- -j1 -v -k0
[0/2] /usr/local/bin/cmake -P /home/runner/work/R-Darktable/R-Darktable/build/CMakeFiles/VerifyGlobs.cmake
[1/824] cd /home/runner/work/R-Darktable/R-Darktable/src && sh /home/runner/work/R-Darktable/R-Darktable/src/tools/create_version_c.sh /home/runner/work/R-Darktable/R-Darktable/build/bin/version_gen.c
fatal: No names found, cannot describe anything.
Version string: 6a628eb
[2/824] cd /home/runner/work/R-Darktable/R-Darktable/build/bin/external && /usr/local/bin/cmake -E make_directory /home/runner/work/R-Darktable/R-Darktable/build/bin/../share/darktable/rawspeed/ && /usr/local/bin/cmake -E copy /home/runner/work/R-Darktable/R-Darktable/src/src/external/rawspeed/data/cameras.xml /home/runner/work/R-Darktable/R-Darktable/src/src/external/rawspeed/data/showcameras.xsl /home/runner/work/R-Darktable/R-Darktable/build/bin/../share/darktable/rawspeed/
[3/824] /usr/bin/g++-8 -DDEBUG -DGDK_DISABLE_DEPRECATED -DGTK_DISABLE_DEPRECATED -DGTK_DISABLE_SINGLE_INCLUDES -D_GLIBCXX_SANITIZE_VECTOR -D_XOPEN_SOURCE=700 -D__GDK_KEYSYMS_COMPAT_H__ -I/home/runner/work/R-Darktable/R-Darktable/build/lib/darktable/rawspeed -I/home/runner/work/R-Darktable/R-Darktable/src/src/external/rawspeed -I/home/runner/work/R-Darktable/R-Darktable/build/lib/darktable/rawspeed/src -I/home/runner/work/R-Darktable/R-Darktable/src/src/external/rawspeed/src/librawspeed -isystem /home/runner/work/R-Darktable/R-Dar [... cut ...]
/home/runner/work/R-Darktable/R-Darktable/src/src/external/rawspeed/src/librawspeed/decompressors/VC5Decompressor.cpp:581:9: error: ‘__PRETTY_FUNCTION__’ not specified in enclosing ‘task’
         assert(!decodedData.has_value() && "Decoded this band already?");
         ^~~~~~
/home/runner/work/R-Darktable/R-Darktable/src/src/external/rawspeed/src/librawspeed/decompressors/VC5Decompressor.cpp:573:9: error: enclosing ‘task’
 #pragma omp task default(none) shared(decodedData, errLog, exceptionThrown)    \
         ^~~
```

LebedevRI, 2022-05-11:
Duplicate of #9988.

LebedevRI, 2022-05-11:
... or to spell it out in more words:
* yes, on `stable` branch, with-asserts build with old GCC does not work. This is a clear GCC bug. I can't workaround it.
* This is not a problem going forward, because in `develop`, as discussed in that issue, LLVM10/GCC10 is required.

aurelienpierre, 2022-05-11:
Ok, sorry for the noise.

closed 2022-05-11 by aurelienpierre