darktable-org/rawspeed #371: Mavic DJI 3 cine, flawed meta data leads to failed import

MMCMA, 2022-06-06:
I am using darktable to import RAW images and I noted issues with importing DJI Mavic 3 DNG files. It seems that they have an error in the meta-data. More details can be found here: https://mavicpilots.com/threads/m3-dng-files-cant-be-imported-into-darktable-3-8-0-error-in-metadata.120986/ . Particulary, the tag "StripByteCounts" seems to cause the problem. 

It seems the problem is fixed with the latest DJI firmware update (early June 2022), however, I still have a lot of "flawed" DNG pictures, I tried to fix them via pyexiv2 and exiv2 but it seems I can't change this tag. Is there a way to add an exception in rawspeed for this case? Or any other suggestions? I tried to hex fix them, but couldn't get it running.

kmilos, 2022-06-07:
This is indeed a known issue: https://github.com/darktable-org/darktable/issues/10852

I'm not aware of another fix apart from correcting the tag in the hex editor, both exiv2 and exiftool treat it as read only.

MMCMA, 2022-06-07:
I replaced it in a hex edit, but get the same error in darktable. However, when I read tag via python it seems correct - but I also see another warning saying "Directory SubImage1, entry 0x0111: Strip 0 is outside of the data area; ignored."  something must go wrong with the hex adjustment. Can you recommend a tool or point me to some documentation what to do?

kmilos, 2022-06-08:
You have to correct the right StripByteCounts as there are multiple ones for multiple images in the file (thumbnails + raw), and make sure you respect the endianness.

```
$ exiv2 -Pxkvh -g StripByteCounts Hasselblad\ -\ L2D-20c\ -\ 16bit\ \(4_3\).DNG
0x0117 Exif.Image.StripByteCounts                    30746
  0000  1a 78 00 00                                      .x..

0x0117 Exif.SubImage1.StripByteCounts                41775360
  0000  00 71 7d 02                                      .q}.

0x0117 Exif.SubImage2.StripByteCounts                930598
  0000  26 33 0e 00                                      &3..
```

Please close this issue as a duplicate and continue discussing on the original issue where some other Mavic user might want to pitch in with their solution, or on the DJI forums, as after all, this is a paid for product that is not functioning correctly and should be supported by the vendor, not by the open source community for free.

As far as documentation goes, I can only point you to the DNG and TIFF specifications:

https://helpx.adobe.com/content/dam/help/en/photoshop/pdf/dng_spec_1_6_0_0.pdf
https://download.osgeo.org/libtiff/doc/TIFF6.pdf