darktable-org/rawspeed #374: Fujifilm X-H2S Support

etrigan63, 2022-07-01:
Please add support for uncompressed, lossless compressed and compressed raw files. I can submit raw files if needed. Just point me to where you want them.

cytrinox, 2022-07-02:
Please upload the samples here: https://raw.pixls.us/

etrigan63, 2022-07-03:
Done!

kmilos, 2022-07-04:
Please try out adding the following to your cameras.xml and report what you find:

```
	<Camera make="FUJIFILM" model="X-H2S">
		<ID make="Fujifilm" model="X-H2S">Fujifilm X-H2S</ID>
		<CFA2 width="6" height="6">
			<ColorRow y="0">BGGRGG</ColorRow>
			<ColorRow y="1">GRBGBR</ColorRow>
			<ColorRow y="2">BGGRGG</ColorRow>
			<ColorRow y="3">RGGBGG</ColorRow>
			<ColorRow y="4">GBRGRB</ColorRow>
			<ColorRow y="5">RGGBGG</ColorRow>
		</CFA2>
		<Crop x="0" y="0" width="0" height="0"/>
		<Sensor black="1024" white="16383"/>
		<ColorMatrices>
			<ColorMatrix planes="3">
				<ColorMatrixRow plane="0">12836 -5909 -1032</ColorMatrixRow>
				<ColorMatrixRow plane="1">-3087 11132 2236</ColorMatrixRow>
				<ColorMatrixRow plane="2">-35 872 5330</ColorMatrixRow>
			</ColorMatrix>
		</ColorMatrices>
	</Camera> 
	<Camera make="FUJIFILM" model="X-H2S" mode="compressed">
		<ID make="Fujifilm" model="X-H2S">Fujifilm X-H2S</ID>
		<CFA2 width="6" height="6">
			<ColorRow y="0">BGGRGG</ColorRow>
			<ColorRow y="1">GRBGBR</ColorRow>
			<ColorRow y="2">BGGRGG</ColorRow>
			<ColorRow y="3">RGGBGG</ColorRow>
			<ColorRow y="4">GBRGRB</ColorRow>
			<ColorRow y="5">RGGBGG</ColorRow>
		</CFA2>
		<Crop x="0" y="0" width="0" height="0"/>
		<Sensor black="1024" white="16383"/>
		<ColorMatrices>
			<ColorMatrix planes="3">
				<ColorMatrixRow plane="0">12836 -5909 -1032</ColorMatrixRow>
				<ColorMatrixRow plane="1">-3087 11132 2236</ColorMatrixRow>
				<ColorMatrixRow plane="2">-35 872 5330</ColorMatrixRow>
			</ColorMatrix>
		</ColorMatrices>
	</Camera> 
```

Probably the "new" lossy compressed mode is not yet supported, like it is the case w/ T4, E4, S10, etc...

etrigan63, 2022-07-04:
No error message, but the image is either blank white or magenta.
[image: image.png]
[image: photo]
Carlos Echenique
Photographer at  CE Photographic
P  305.219.2433  <305.219.2433>
E  ***@***.***  ***@***.***>
W  cephotographic.com  <http://cephotographic.com>
<http://instagram.com/cephotographic> <http://facebook.com/cephotos>
<http://www.flickr.com/photos/echenique>

"A day without sunshine is like, you know, night." - Steve Martin.
Create your own email signature
<https://www.wisestamp.com/create-own-email-signature/?utm_source=promotion&utm_medium=signature&utm_campaign=create_your_own&srcid=>

kmilos, 2022-07-04:
I had already updated the settings above in the meantime, please give it a go again.

etrigan63, 2022-07-04:
> I had already updated the settings above in the meantime, please give it a go again.

This one is much better. Colors are correct. Filmic RGB underexposes an additional stop.

etrigan63, 2022-07-04:
> Only two samples uploaded, but H2S supports 3 modes (uncompresses, lossless, compressed). Please uploaded the missing sample, too.

Lossy compressed sample uploaded.

mwperina, 2022-09-30:
Hi,
I've tried to append above code into cameras.xml, but when I'm trying to open uncompressed RAF file in Darkroom, following error is shown:

```
`Fujifilm X-H2S' color matrix not found!
```

I'm using Darktable 3.8.1 on Fedora 35. 

Is there anything I can help with to include X-H2S support into Darktable?

Thanks,
Martin

etrigan63, 2022-10-04:
The XML provided no longer works as of dt 4.0.1. The X-H2S firmware updated to 2.0 but i don't think they changed the file format.

mwperina, 2022-10-11:
After upgrade to Darktable 4.0.1 I can import uncompressed RAF from X-H2S without any problem and after working on more than 50 photos I haven't found any obvious issue, images are processed fine. So is there anything blocking us from getting X-H2S support merged?

closed 2022-11-18 by LebedevRI