darktable-org/rawspeed #376: Some different black points at different ISO, Canon EOS D60

PeterWem, 2022-07-25:
```
<Camera make="Canon" model="Canon EOS D60">
		<ID make="" model="">Canon EOS D60</ID>
		<Sensor black="132" white="4000"/>
		<Sensor black="137" white="4000" iso_list="800"/>
	</Camera>
```
[image]
```

exiv2 -pt CRW_7102.dng 2>/dev/null | grep ColorMatrix2
Exif.Image.ColorMatrix2                      SRational   9  6211/10000 -1358/10000 -896/10000 -8557/10000 15766/10000 3012/10000 -3001/10000 3507/10000 8567/10000
```

[image]

PeterWem, 2022-11-19:
The optical black area in rawspeed is wrong why the picture will look green at high ISO. I will correct it with a PR instead.

closed 2022-11-19 by PeterWem

LebedevRI, 2022-11-19:
If they are, please feel free to PR them.

But the whole of "black 'area'" handling is completely broken (#389, etc).