darktable-org/rawspeed #381: Sony ILCE-7M4 lossless compressed images not supported

LucasRueckert, 2022-09-07:
Hi,
I own a Sony ILCE-7M4 and I'd like to use the lossless compressed raw format.
In this issue [#12064](https://github.com/darktable-org/darktable/issues/12064) it was hinted that support in rawspeed was necessary.
I also uploaded some sample images to [raw.pixls.us](raw.pixls.us)

I'd like to help to get this working, so if somebody could point me to some documentation would be nice.

lclevy, 2022-09-07:
Hi, for your information, I started documenting Sony raw formats here: https://github.com/lclevy/sony_raw

kmilos, 2022-09-07:
Duplicate of https://github.com/darktable-org/darktable/issues/10556 and https://github.com/darktable-org/darktable/issues/8523, please close.

@lclevy The Sony lossless format is understood (4-channel JPEG SOF3).

lclevy, 2022-09-07:
@kmilos any reference, please?

kmilos, 2022-09-07:
I added the link to my analysis.

kmilos, 2022-09-07:
@lclevy I'm not aware of any other "formal" reference when it comes to the Sony lossless mode - you can maybe as background point to the JPEG itu-t81.pdf spec itself for the SOF3 (LJPEG92) description and then to the Adobe DNG SDK for how it's used slightly differently in DNGs (2 channels instead of 4), also described here: https://thndl.com/how-dng-compresses-raw-data-with-lossless-jpeg92.html

NickLplus, 2022-09-07:
> @kmilos any reference, please?

Here is LibRaw implementation:
https://github.com/LibRaw/LibRaw/blob/master/src/decoders/decoders_libraw_dcrdefs.cpp#L75

closed 2022-09-08 by LucasRueckert