darktable-org/rawspeed #384: Only partial support for OM Systems OM-1 images

greymont, 2022-09-17:
I was thrilled to learn that darktable 4.0.1 was out! I downloaded the 64bit Windows version and installed it over my previously installed 4.0 (from a special build supporting the OM Systems OM-1). When I ran it, initially I saw lots of thumbnails. But as I scrolled to more images from last month, I saw more and more skulls! Attached is a screenshot.

I cannot understand why. The image data from the skulls looks the same as for the images with thumbnails. Same camera, same lens. These are not high res images vs. normal res images. Seems like once supported, they should all be processed equally. Can anyone explain this?

[image]

kmilos, 2022-09-20:
This camera is currently not supported by rawspeed (nor the last released exiv2 version), so there is no point researching the issue at this time. Furthermore, thumbnail generation/extraction and lighttable view fall more under darktable responsibility. Please refer to whomever provided the custom build.

See https://github.com/darktable-org/darktable/issues/11258, https://github.com/darktable-org/rawspeed/pull/348 and https://github.com/darktable-org/rawspeed/pull/357

sarunasb, 2022-11-02:
@greymont Your screenshot is of darktable 4.0.1, not the “special” 4.0.om1 build. Standard, downloaded version of darktable does not support OM System yet. Thumbnails which are not skulls might be from cache.

greymont, 2022-11-26:
Can anyone confirm that OM-1 support is now in the rawspeed version that will be used with the next darktable release? I've been poking around and cannot find any details on this (not familiar enough with Github).

ptilopteri, 2022-11-26:
googling "darktable rawspeed" provides https://github.com/darktable-org/rawspeed
which shows "cameras:929" and is a link to the rawspeed *currently* supported cameras.

greymont, 2022-11-26:
> googling "darktable rawspeed" provides https://github.com/darktable-org/rawspeed which shows "cameras:929" and is a link to the rawspeed _currently_ supported cameras.

And yet nothing I can find there answers the question. Yes, sample RAW files were submitted about 10 months ago. Yes, a merge request was submitted months ago but has not been accepted yet and is still listed as needing a look by a committer. Yes, there is an outstanding question from the submitter as to why this has not been accepted but no answer. Hence my question.

LebedevRI, 2023-02-04:
The support was merged in #357 / #399, but not propagated into darktable yet.

closed 2023-02-04 by LebedevRI

kmilos, 2023-02-04:
Btw, requires exiv2 0.27.6 for the thumbnails to work on import.