darktable-org/rawspeed #391: rawspeed (and so darktable) should read levels from the raw

aquatix, 2022-10-23:
When editing my photo's in DT, I came across the issue where the whitepoint being used wasn't the one that's in the CR2, and which exiftool shows.

All cr2 should read levels from the raws, here's an example:

https://discuss.pixls.us/t/struggling-with-direct-light/33199

According to exiftool the raw whitepoint should be 11765 but DT sets it to 13653

I'm running Ubuntu 22.10 with Darktable 4.0.1, but another user had the same issue with DT 4.1 from master

AxelG-DE, 2022-10-23:
Here is "the other user" :-P

Experienced that through above pic as with the initial values, the raw-overexposed indicator does not show obviously blown highlights.

kmilos, 2023-11-24:
Closing in favor of https://github.com/darktable-org/rawspeed/issues/417

closed 2023-11-24 by kmilos