darktable-org/rawspeed #400: The change to needing cmake 3.18 prevents building darktable under Ubuntu 20.04

RawConvert, 2022-11-18:
Ubuntu is of course one of the most widely used distros, if not The most used, and 20.04 is long-term supported, which makes it attractive to many users.  Rawspeed has recently changed and now needs cmake version 3.18, whereas 20.04 only has cmake 3.16.3.
Is it really necessary to move to 3.18 and please please, if it's possible, can it go back to where it was (or 3.16.3 say)?

I just searched this repository for "cmake 3.18" hoping to understand why it's been advanced but only one item was found and there seems to be no info about why upgraded.

[image]

Thanks

kmilos, 2022-11-18:
@LebedevRI FYI, this didn't show up straight away w/ the darktable CI because (fortunately or not) the [Ubuntu 20.04 GHA runner has a more recent CMake installed in addition.](https://github.com/actions/runner-images/blob/main/images/linux/Ubuntu2004-Readme.md)

LebedevRI, 2022-11-18:
> @LebedevRI FYI, this didn't show up straight away w/ the darktable CI because (fortunately or not) the [Ubuntu 20.04 GHA runner has a more recent CMake installed in addition.](https://github.com/actions/runner-images/blob/main/images/linux/Ubuntu2004-Readme.md)

Yeah, not surprised.
This is why i always try not to use the provided github image,
but do `container: image: <name of barebones debian ubuntu docker image>`,
so there's no confusion over what is and isn't there.

closed 2022-11-18 by LebedevRI