darktable-org/rawspeed #403: Camera support for ExpertRAW (Samsung S22 Ultra)?

PeterWem, 2022-11-21:
ExpertRAW from Samsung seems to be already demosaiced like Apple ProRAW. darktable gives the following error:

[image]
ExpertRAW samples uploaded to raw.pixls.us and also here:

[20221120_114027_UltraWide.zip](https://github.com/darktable-org/rawspeed/files/10055013/20221120_114027_UltraWide.zip)
[20221120_114038_Wide.zip](https://github.com/darktable-org/rawspeed/files/10055020/20221120_114038_Wide.zip)
[20221120_114044_Tele.zip](https://github.com/darktable-org/rawspeed/files/10055023/20221120_114044_Tele.zip)
[20221120_114100_SuperTele.zip](https://github.com/darktable-org/rawspeed/files/10055025/20221120_114100_SuperTele.zip)

kmilos, 2022-11-21:
Not the same error as Apple's ProRAW (which is clearly reported to do w/ currently unsuppored LJpeg predictor), so something else is going on...

LebedevRI, 2022-11-21:
Please contribute all relevant samples to https://raw.pixls.us/
I'm guessing here only a single sample will be needed.

kmilos, 2022-11-21:
Looks like there is a DRI segment (restart markers) that is currently not supported?

kmilos, 2022-11-21:
Btw, this is pretty weird as well for a demosaiced raw (someone should let "samsung" know) - might create issues later even when the decoding is made to work:

```
BlackLevelRepeatDim             : 2 2
BlackLevel                      : 0 0 0 0 0 0 0 0 0 0 0 0
```

PeterWem, 2022-11-21:
> Please contribute all relevant samples to https://raw.pixls.us/ I'm guessing here only a single sample will be needed.

They should already be there.

LebedevRI, 2022-11-21:
Well, i asked because i can not find them there, including unreviewed submissions.
Under which camera name are they there?

PeterWem, 2022-11-21:
Should be Samsung SM-S908U  Galaxy S22. I will tey a sample again.

LebedevRI, 2022-11-21:
Thanks, they were there under `Samsung SM-S908U`, i looked for `S22`...
I've now explicitly disabled support for LJPEG with restart interval != 0.
We could probably support this, i'm just worried about pessimising the performance
of the rather dominant case of no restart intervals...

kmilos, 2022-11-21:
> I've now explicitly disabled support for LJPEG with restart interval != 0.
> We could probably support this, i'm just worried about pessimising the performance
> of the rather dominant case of no restart intervals...

Yeah, but probably these will not go away and will start appearing in more and more Samsung models, like Apple's ProRAW... There's always the Adobe DNG Converter route of course...

zOrg1331, 2023-05-01:
Hi, here is what I receive with Samsung Galaxy S22 DNG files on the current master branch:
```
     6.5248 [rawspeed] (20230427_113626.dng) void rawspeed::AbstractDngDecompressor::decompress() const, line 218: Too many errors encountered. Giving up. First Error:
static void rawspeed::AbstractLJpegDecoder::parseDRI(rawspeed::ByteStream), line 274: Non-zero restart interval not supported.
     6.5248 [libraw_open] extensions whitelist: `cr3'
```
What would be helpful from me to resolve the issue?

jirislav, 2023-07-10:
I'm experiencing the same issue on my Samsung Galaxy S23 (non-Ultra), and I'd love to use Darktable. I've noticed there hasn't been any progress on this issue since last November.

> We could probably support this, i'm just worried about pessimising the performance of the rather dominant case of no restart intervals...

@LebedevRI I understand your concern about potential performance degradation for the most common cases where there are no restart intervals. However, if the program crashes during DNF loading, thus disabling Darktable's darkroom functionality, performance doesn't really matter.

Could we consider some potential trade-offs?

- For instance, could we only enable restart intervals for LJPEG when a crash occurs as a recovery method? I understand that this might slow down the process significantly, and as @kmilos suggested, using the `Adobe DNG Converter` might be a more effective approach.
- Alternatively, could we make this a configurable option in Darktable's settings? We could include a warning stating that enabling this option could lead to potential performance degradation, especially in the 75% of cases where the feature would otherwise work perfectly.
- Or we could possibly come to a solution that would branch the logic dynamically without much of a performance cost? As @kmilos pointed out, this use-case might become very common in the near future. 

In any case, I'd love to contribute a Pull Request if we first come up with a preferred solution here, so I would welcome any suggestions on where to make the changes as I'm new to this project.

jirislav, 2023-07-10:
Just a thought - couldn't we leverage restart intervals to load the image in parallel, thus making it even faster than normal?

LebedevRI, 2023-07-10:
> Just a thought - couldn't we leverage restart intervals to load the image in parallel, thus making it even faster than normal?

Theoretically -- sure. In practice the gains will be offset by the fact
that you've now doubled the amount of memory reads.
I guess the main factor will be the ratio between the input row pitch vs CPU data cache sizes/layout.

LebedevRI, 2023-07-10:
> I'm experiencing the same issue on my Samsung Galaxy S23 (non-Ultra), and I'd love to use Darktable. I've noticed there hasn't been any progress on this issue since last November.
>
>
> @LebedevRI I [...]

I appreciate the thought, but this either needs to be supported properly, or not at all.

kiss81, 2023-07-21:
No luck with the s23 and darktable 4.4.1

jirislav, 2023-07-21:
> I appreciate the thought, but this either needs to be supported properly, or not at all.

Could you please describe, what would the proper solution look like? And do you have any suggestion on where to start to make the things move forward?

rathbone1200cc, 2023-08-02:
I see the same error with a Samsung S23 Ultra and darktable 4.4.2.

As an aside, I can also confirm that converting the DNG files via `Adobe DNG Converter` works around the error.  Sadly `Adobe DNG Converter` doesn't run natively on linux, requires `wine`, runs slowly, etc. so this remains an inconvenience.

Thanks for tracking this.  I hope to see support for these Samsung DNGs when the darktable contributors can prioritize this issue.

LebedevRI, 2024-01-27:
^_^

[image]

closed 2024-01-28 by LebedevRI