darktable-org/rawspeed #405: Sony A74 (ILCE-7M4) support

nugecom, 2022-11-21:
[Re-posting after error putting this as Pull Request]

Hi,

I couldn't find this issue in previous issues so here goes:-

    Sony A7iv camera
    Uncompressed RAW files (.arw)
    darktable: 4.0.0
    Ubuntu linux 20.04

In darktable, which uses rawspeed, some of my photographs are showing blown highlights as a 'lilac' colour, rather than white as I would have expected. I have a work around for now, but...

Is this a possible rawspeed issue or a darktable issue?

Also, I am unable to process lossless compressed files in darktable and was unsure if this was because rawspeed does not yet support Sony A74 (ILCE-7M4). If rawspeed does not, do you have a timeline when it might? If it does, is there a bug in rawspeed or perhaps darktable?

Your help is greatly appreciated! Thanks,

John

LebedevRI, 2022-11-21:
The highlights issue should have been fixed but the fix has not been propagated to darktable yet.
As for lossless compressed files, they will be supported once they are supported. See existing bugs.
It looks like there's an open pull request here that claims to add that support, so once i get to it i will look into it.

closed 2022-11-21 by LebedevRI

nugecom, 2022-11-21:
Thanks Roman. 

Are you aware of any other issues with Sony A7iv raw files from your end? I seem to be correcting for a slight green tint more often than I'd expect. I recently moved from a Nikon D750 which seemed more colour accurate.

LebedevRI, 2022-11-21:
I'm sorry, i'm not familiar with those cameras.
It could be wrong white balance, it could be that the default camera white balance is calculated differently,
it could be bad color matrix (the one we got from Adobe DNG Converted), it could be that the lighting under which
you are operating requires different color matrix, etc, etc, etc.

starsforeveryone, 2022-12-17:
> As for lossless compressed files, they will be supported once they are supported. See existing bugs.
> It looks like there's an open pull request here that claims to add that support, so once i get to it i will look into it.

Could this be added to the upcoming christmas release? 🥺

Thanks in advance!

TurboGit, 2022-12-17:
> Could this be added to the upcoming christmas release? sob

No, way too late now, a release is not done in a single day. The code is frozen since many days now.

LebedevRI, 2022-12-17:
FWIW, i'm very intentionally ignoring all but simple `cameras.xml` PR's currently.
I totally can go into PR frenzy, since dt freeze does not mean rawspeed is frozen,
but if i do merge something, i have to //consider// how it affects the downstreams.
Note the word consider, it is meant in the sense consult, not "be dictated terms".

starsforeveryone, 2022-12-17:
Thanks for answering. I understand the situation. Would it be possible to add the code for lossless compressed raw support after the release?