darktable-org/rawspeed #417: Get black level and white point from EXIF for Canon

afranke, 2023-01-11:
Something similar to #385 should be done for Canon cameras to replace the per-model, per-ISO table in [data/cameras.xml](https://github.com/darktable-org/rawspeed/blob/develop/data/cameras.xml). https://exiftool.org/TagNames/Canon.html#ColorData4 has the relevant info (`SpecularWhiteLevel` and `AverageBlackLevel` are what we are after).

Happy to help with testing or providing data from my library of photographs, but I can’t submit a PR.

For what it’s worth, a search on “canon” currently yields 16 open issues in this project, and it looks like most if not all would be solved by implementing this.

(This is a followup from darktable-org/darktable#13320)

saknopper, 2023-01-11:
I think the hard thing (and why nobody has done it yet...) about doing this for Canon cameras is the fact that the tags to be used differ a lot between camera models. In other words, the ColorData4 tags you mention are only valid for a limited set of cameras.

So getting this to work across all Canon cameras requires a bit of effort.

kmilos, 2023-01-11:
There's no reason not to try to read all of the currently (Exiftool or otherwise) known locations in ColorData4/5/6/7/8/9/10/11

A bit of effort, yes, but not impossible - mostly "ugly"... So, waiting for a motivated Canon vested contributor ;)

afranke, 2023-01-11:
:thinking: doesn’t the `ColorDataVersion` field (and its presence/absence) tell us where to read the other fields though?

PeterWem, 2023-01-20:
You have exif tags for black level from different Canon models in this thread https://www.dpreview.com/forums/thread/3789550

I believe getting black level from optical black area is still better than reading from Exif, as long as the right fields in the black area are choosen. That has been a problem, because at ISO 100 all fields look black but at higher ISO some fields are not black anymore.

A new thing is also Canon CRAW that trashes the optical black area. At least it looks like that just viewing the fields. Though I haven't seen any difference when darktable choose black level. 

I would like to be able to choose if now reading from Exif make it through. Perhaps something similar as the new lens correction in darktable (reading from Lensfun or Exif).

Sample files when it went wrong and when the black area above wasn't black https://discuss.pixls.us/t/measure-black-and-white-levels/32082/11?u=peter