darktable-org/rawspeed #422: Move bit depth from mode to sensor data?

twbaumann, 2023-02-02:
Just an idea: Why can't we add an optional bit_depth attribute to the sensor data in cameras.xml - next to the black and white levels, similar to iso_min and iso_max. Then the bit depth could be removed from the many Nikon camera modes which would prevent a lot of duplicate data, making the file cleaner and less error prone.

I did a comparison of all entries that only differ in bit size of mode. As expected the only differences lie in the sensor black and white levels in combination with optional iso ranges, with one exception: The Nikon D800(E) crop parameters for uncompressed modes ("7424x4924-14bit-uncompressed" _cropped_, "14bit-uncompressed" _not cropped_, "12bit-uncompressed" _cropped_) which seems to be inconsistent anyway.

The selection of sensor data in Camera::getSensorInfo(int iso, int bit=0) could be done in a backwards compatible way like this:
1. Find entry matching iso range and bit_depth
2. Find first entry matching iso range or bit_depth, the other being missing
3. Take the default entry with both iso range and bit_depth missing

That would imply changing Camera.\*, CameraSensorInfo.\*, NefDecoder.\* and possibly some base class - of course in a way that does not affect the other converters. 

Perhaps I am missing some drawback. What do you think?

LebedevRI, 2023-02-03:
Can we simply read those levels from the EXIF instead?

twbaumann, 2023-02-03:
I am afraid not, a random check of Nikon D3 and D5 samples from raw.pixls.us with exiv2 did not reveal those level data.

LebedevRI, 2023-02-04:
I think the main questions are: "what are we trying to fix?", "what are we trying to solve?",
and, "is that a sufficiently general solution?".

I agree that the current `mode` is ugly, but moving `bitdepth` elsewhere still retains the rest of the mode.
And all that churn just to prettify something that should not exist in the first place seems like a not-good trade.

twbaumann, 2023-02-04:
Data deduplication and a perhaps more logical place for bit depth in the xml file was my motivation. But you are right, it does not fix anything and in addition it is Nikon specific. Just close the issue.

closed 2023-03-21 by LebedevRI