darktable-org/rawspeed #423: Sony ILCE-7M4 (A7IV) Lossless Raw support missing

dtrtuser, 2023-02-04:
Hi,

Latest official version supports ILCE-7M4 lossy and uncompressed raw but doesn't support Lossless Raw. I've seen that sample of lossless compressed raw files have been uploaded in the RawSpeed repository but if needed I can upload more.

Cheers.

kmilos, 2023-02-04:
This is a known issue (the release notes explicitly mention it), and therefore, a duplicate report. Please close and make use of the search facility before opening new issues.

See https://github.com/darktable-org/rawspeed/issues/381 https://github.com/darktable-org/rawspeed/issues/328 https://github.com/darktable-org/darktable/issues/10556 and https://github.com/darktable-org/darktable/issues/8523

dtrtuser, 2023-02-04:
Closing this issue as requested.

closed 2023-02-04 by dtrtuser