darktable-org/rawspeed #429: Color-space after ColorMatrix transformation

gustavolsson, 2023-02-16:
Hello and thanks for creating and maintaining rawspeed! I'm using the library to create a simple raw editor but I find it difficult to figure out some things:

1. Is the raw pixel data we get unprocessed sensor data (uniformly scaled to appropriate ranges)? This is evident if there is a CFA because the monochrome values need to be manually converted to the nondescript "RED" "GREEN" and "BLUE" channels, but is it also the case for non-CFA images?

2. I assume that the ColorMatrix from the camera metadata is never applied by rawspeed?

3. What color-space is the RGB values in after a transformation by the ColorMatrix metadata value?

4. Are the white balance metadata coefficients supposed to be used as-is or are they in a particular color-space? I.e. do I need to convert the RGB values into e.g. [LMS color space](https://en.wikipedia.org/wiki/LMS_color_space) before applying the scaling as [outlined here](http://www.brucelindbloom.com/index.html?Eqn_ChromAdapt.html)?

I think it would help to document each metadata value in the `data/README.md` file. I'm open to doing this with some guidance!

LebedevRI, 2023-02-16:
> 1. Is the raw pixel data we get unprocessed sensor data (uniformly scaled to appropriate ranges)?

It depends on your definition of "unprocessed" and "scaled", but basically yes,
library does not touch the data in ways that are not strictly required for decoding itself.
It doesn't even normalize the data.

> This is evident if there is a CFA because the monochrome values need to be manually converted to the nondescript "RED" "GREEN" and "BLUE" channels, but is it also the case for non-CFA images?

Notionally - yes.

> 2. I assume that the ColorMatrix from the camera metadata is never applied by rawspeed?

Correct.

> 3. What color-space is the RGB values in after a transformation by the ColorMatrix metadata value?

That color matrix is effectively taken from DNG's `ColorMatrix2`, and is notionally for illuminant `7` (D65),

> Description
> ColorMatrix2 defines a transformation matrix that converts XYZ values to reference camera
> native color space values, under the second calibration illuminant. The matrix values are
> stored in row scan order.

> 4. Are the white balance metadata coefficients supposed to be used as-is or are they in a particular color-space? I.e. do I need to convert the RGB values into e.g. [LMS color [...]

I'm not familiar with chromatic adaptation stuff. Normally, you'd just use them as-is.

LebedevRI, 2023-02-21:
I'm going to close this.
If you have thoughts on how the docs could be improved, please feel free to PR that.

closed 2023-02-21 by LebedevRI

kmilos, 2023-02-23:
> or if the source data is in the original raw file itself

Most raw formats don't provide this information.

LebedevRI, 2023-03-21:
> Do you know if this data comes from Adobe's in-house camera calibration (encoded into the Adobe DNG converter tool) or if the source data is in the original raw file itself?

I believe, //some// non-DNG raw formats do contain color matrix in the metadata,
but i don't think we read it. But i'm not sure if that is used in ADC, or what the source for ADC matrices is.