darktable-org/rawspeed #480: Crop setting for Olympus M1 Mark III

neutrino82, 2023-06-02:
I just checked the camera.xml file for my camera (Olympus M1 MarkIII). No crop is set there for the ORF raw file. As such when opening ORF files using darktable I get a full 5240x3912 image. However on some (not all) of the shots an artifact in the form of a 32 px wide dark band is present along the right edge of the frame. Could a set like <Crop x="0" y="3" width="-32" height="-3"/> solve the issue?

LebedevRI, 2023-06-02:
Could you please share one affected image here?
zip it and upload, if filesize allows...

neutrino82, 2023-06-02:
Here is an example. I attached the ORF file.

I cannot clearly address which crop is applied in camera. A brute comparison seems to indicate 12px on the top and bottom, 44 on the right and 12 on the left so that one could think of somthing like: <Crop x="12" y="12" width="-44" height="-12"/> but I suppose that crop is applied after lens correction on the JPEG and the darker line (at least on my camera is just 32 px on the right. Maybe something like <Crop x="0" y="3" width="-32" height="-3"/> would be enough to avoid the dark line and preserve the 4:3 ratio.
[P5130026_ORF.zip](https://github.com/darktable-org/rawspeed/files/11634337/P5130026_ORF.zip)

paolodepetrillo, 2023-06-03:
I have the same issue with the E-M1 II, exact same pixel dimensions.

The ORF file contains the tags that indicate how it should be cropped:
```
Exif.OlympusIp.CropLeft                      Short       2  12 0
Exif.OlympusIp.CropTop                       Short       2  12 0
Exif.OlympusIp.CropWidth                     Long        1  5184
Exif.OlympusIp.CropHeight                    Long        1  3888
```

If it's converted to DNG with Adobe DNG Converter, it puts those values in the DefaultCropOrigin and DefaultCropSize tags, and darktable sees those and automatically applies them, hiding the dark band on the right side. The resulting image does have the exact 4:3 ratio after applying these crops.

neutrino82, 2023-06-03:
Thanks. So based on that it should translate in a

`<Crop x="12" y="12" width="-44" height="-12"/>`

in the camera.xml here (as far as i understand)

LebedevRI, 2023-06-03:
@paolodepetrillo please open a new issue and similarly attach an affected sample.

paolodepetrillo, 2023-06-03:
The cause of the artifact is the camera's Shading Compensation (vignetting correction) function. It looks like the correction doesn't get applied to that 32 pixel strip on the right. So the artifact is only visible if:

- Shading Compensation is On in settings
- The lens supports vignetting correction
- The correction is strong enough to be noticeable at the current zoom level / focus distance

LebedevRI, 2023-06-03:
> so I feel in this case ones expect to classify as junk just the 32px on the right ending up with:
> `<Crop x="0" y="3" width="-32" height="-3"/> `

Yep, `width="-32"` does appear like the right thing.
I don't really see any bad horizontal rows, so i'm not sure why we'd need to crop top/bottom.

closed 2023-06-03 by LebedevRI

neutrino82, 2023-06-03:
Yes. I just hypotesize that additional crop to preserve the 4:3 aspect ratio in the imported image.

LebedevRI, 2023-06-03:
Right. I don't think we generally do that.