darktable-org/rawspeed #491: src/librawspeed/README.md is out of date

mallman, 2023-06-30:
I recently attempted to update my version of rawspeed and found that certain APIs I used before have changed. Some I don't know how to replace. For example, I used to use the `RawImageData->getData()` and `RawImageData->getData(int, int)` apis, but those are gone. I don't know how to update my code. Would you please go through that file and update it for usage with the current API?

Thank you.

LebedevRI, 2023-06-30:
Yup, docs are not up to date.

> For example, I used to use the `RawImageData->getData()` and `RawImageData->getData(int, int)` apis, but those are gone. I don't know how to update my code.

You want `get{U16,F32}DataAs{Uncropped,Cropped}Array2DRef()`, which returns a thin 2d view.

mallman, 2023-07-02:
> Yup, docs are not up to date.
> 
> > For example, I used to use the `RawImageData->getData()` and `RawImageData->getData(int, int)` apis, but those are gone. I don't know how to update my code.
> 
> You want `get{U16,F32}DataAs{Uncropped,Cropped}Array2DRef()`, which returns a thin 2d view.

Hi @LebedevRI. Thank you for the guidance. I could use a little more help.

The API I'm integrating with requires a linear `uint16_t*` buffer (of the decoded raw data), similar to what `RawImageData->getData()` provided. Is it still possible to get the underlying `uint8_t*` or `uint16_t*` from the new API?

If not, what would you recommend as the most efficient way to alloc and populate a `uint16_t*` from the values returned by `getU16DataAs{Uncropped,Cropped}Array2DRef()`?

Thank you.

LebedevRI, 2023-07-02:
See e.g. https://github.com/darktable-org/darktable/blob/6cfcb0b5344a75d2b51e683abbe0c5bd8ffcbda2/src/imageio/imageio_rawspeed.cc#L395-L405
Just taking address of the first column gives that address.

mallman, 2023-07-03:
Thank you. This has helped very much.