darktable-org/rawspeed #497: Looking for bits per sample

mallman, 2023-07-06:
Is there a RawSpeed API for getting a camera's bits per sample (e.g. 12, 14, 16)? I know there's the tiff tag for it, bit there are still some special cases where RawSpeed has to employ some special logic. For example, there are the `PANASONIC_BITSPERSAMPLE` and `FUJI_BITSPERSAMPLE`, among other things. It would be helpful for the API to expose a uniform `bitsPerSample` value that hides all the messy details.

LebedevRI, 2023-07-06:
> Is there a RawSpeed API for getting a camera's bits per sample (e.g. 12, 14, 16)? I know there's the tiff tag for it, bit there are still some special cases where RawSpeed has to employ some special [...]

There is not.I recall having this problem when implementing deflicker in darktable some time ago.

Essentially, what do you want to get? What question would that data help answer?

mallman, 2023-07-06:
> There is not.I recall having this problem when implementing deflicker in darktable some time ago.
>
> Essentially, what do you want to get? What question would that data help answer?

Thanks for asking! I use it as a heuristic for computing the bin width or bin count of a raw histogram. For example, for a 12-bit per sample raw image and a raw histogram with 1024 bins, I infer a bin width of (2^12) / 1024 == 4. Conversely, for a histogram with bin width 4, I infer the number of bins in my raw histogram is (2^12) / 4 == 1024.

Generally, bin width = (2^bps)/(bin count) and bin count = (2^bps)/(bin width).

This is (presently) the only place I use this information.

mallman, 2023-07-06:
> Let's explore Sony ARW2.3 format:
> 
> * 11 bit base values
> * 7-bit delta encoding
> * expansion curve with ~17000 upper value limit
> 
> What is BitsPerSample for this format?

For my purposes, I would have to get my hands on one of these files and play with it. But at first, I would look at what `libraw_colordata_t.raw_bps` returns.

LibRaw, 2023-07-07:
> For my purposes, I would have to get my hands on one of these files and play with it. But at first, I would look at what `libraw_colordata_t.raw_bps` returns.

Just checked:  raw_bps=14 for file(s) in question (ARW2.3) while actual data range is a few more.

mallman, 2023-07-11:
> There is not.I recall having this problem when implementing deflicker in darktable some time ago.

Is this something you'd be okay having as part of the API? If so, I'm likely to attempt a PR, as this is useful information for me.

LebedevRI, 2023-07-11:
What would be the actual definition behind that data? (i.e. how do we decide that it's this value and not the other)
What happens for floating-point data?

mallman, 2023-07-11:
> What would be the actual definition behind that data? (i.e. how do we decide that it's this value and not the other)
What happens for floating-point data?

For raw files encoded with the uint16 component data type, I think of the "bits per sample" representing an upper bound on what a camera advertises as its actual pixel encoding range. So while the raw data itself may be provided in a 16-bit format, the camera only uses the first 14-bits for encoding image data.

I know that's not exactly a precise definition. I'm not sure what it would be _per se_. I know from experience that certain cameras advertise a value in TIFF tag 258, described at https://exiv2.org/tags-xmp-tiff.html. That TIFF tag represents a bits-per-sample per pixel component. (Of course, in RAW that's just one component per pixel).

From my experience, for traditional (non RAW) image files like JPEG and TIFF, that tag will be the actual size of the data type used for encoding. Hence, it will be 8 for JPEG and 8-bit TIFF, and 16 for 16-bit TIFF, etc.

Also, from my experience with Nikon DSLRs and the Sony ILCE-1, the value of that tag is not 16. It's 12 for older cameras and 14 for newer ones. So I'd look to the value of that tag first.

RawSpeed has its own use and definition of bits per sample in various places in its source code. I would look to that value as another option for a public "bitsPerSample" value. I would have to take a closer look at this internal value though.

I don't have experience with 32-bit RAW files. Are there any cameras that produce that format? I understand it's supported by DNG. In either case, I'm supposing the "bits per sample" would be the full 32-bits unless the TIFF tag says otherwise, or if there's some special case with a camera.

LebedevRI, 2023-07-11:
I'm mainly asking because i'm vary of adding something
very obscure and under-specified to the public interface.
(even thought there are no guarantees about the stability of the former)

Would the following definition make sense:
> dataBitDepth
>
> Specifies the "precision" of a sample. Originally, defined as
> the bit depth of the Camera's Analog-to-Digital converter,
> and may be further lowered if bit depth reducing compression,
> was applied.

mallman, 2023-07-11:
> I'm mainly asking because i'm vary of adding something very obscure and under-specified to the public interface. (even thought there are no guarantees about the stability of the latter)
> 
> Would the following definition make sense:
> 
> > dataBitDepth
> > Specifies the "precision" of a sample. Originally, defined as
> > the bit depth of the Camera's Analog-to-Digital converter,
> > and may be further lowered if bit depth reducing compression,
> > was applied.

I like the ADC-based definition. I want to do some more reading on that.

I did some googling and found a relevant forum thread: https://exiftool.org/forum/index.php?topic=9679.0.

Apparently exiftool reports a full 16-bits for BitsPerSample for Canon raw files (or at least the files the OP had). I'm not sure exactly where exiftool gets this information from. I can do some more digging on that, but it's probably just reading the tiff tag.

At the end of the thread, someone reports that Adobe apps use the WhiteLevel tag to derive bits per sample. They use the formula:

```
bits per sample = ceil(log2(WhiteLevel))
```

Apparently this reports a more "accurate" value for bits per sample for canon raw files. I don't know if this works for other manufacturers as well, but I will test it with my Nikon and Sony files.

Anyway, I think this subject bears more investigation on my part. I see this is becoming more complex than anticipated, but I'll do some more digging and report back. I'm still optimistic we can settle on a useful definition with a useful value.

LebedevRI, 2023-07-11:
Yeah, i think you now see my point here :/
If you want something quick-n-dirty, 

> bits per sample = ceil(log2(WhiteLevel))

is indeed what i would expect to use and to just work in most of the cases.
But  if something more precise and realistic is _needed_, things get complicated.

mallman, 2023-07-28:
> Yeah, i think you now see my point here :/ If you want something quick-n-dirty,
> 
> > bits per sample = ceil(log2(WhiteLevel))
> 
> is indeed what i would expect to use and to just work in most of the cases. But if something more precise and realistic is _needed_, things get complicated.

This is working for me:

```c
(uint16_t)ceil(log2(rawImage->whitePoint));
```

I'll close this issue.

Thanks again for your help.

closed 2023-07-28 by mallman