darktable-org/rawspeed #498: Using the ColorFilterArray API to get the filter color for a row/col

mallman, 2023-07-06:
I want to use the `ColorFilterArray` API to get the filter color associated with a given row/col in the raw data 2d array. I want to do this for each of the two raw data representations: `getU16DataAsCroppedArray2DRef()` and `getU16DataAsUncroppedArray2DRef()`. Given a row/col specification into one of either of these arrays, how do I get the `CFAColor`?

LebedevRI, 2023-07-06:
So, i think this is pretty broken, and i didn't touch that yet :)
https://github.com/darktable-org/rawspeed/issues/30 is relevant.

The TLDR is, for X-Trans, the CFA is specified for **un**cropped image,
while for other sensor types, it is specified for **cropped** image.

Does that answer your question?

mallman, 2023-07-07:
> So, i think this is pretty broken, and i didn't touch that yet :) #30 is relevant.
> 
> The TLDR is, for X-Trans, the CFA is specified for **un**cropped image, while for other sensor types, it is specified for **cropped** image.
> 
> Does that answer your question?

Should I be using the `shiftRight()`/`shiftDown()` methods or just `getColorAt(col, row)`?

LebedevRI, 2023-07-07:
Whatever works for you i suppose?

mallman, 2023-07-07:
> Whatever works for you i suppose?

I think I have what I need. Thank you, @LebedevRI.

closed 2023-07-07 by mallman