darktable-org/rawspeed #506: DNG opcode level 2 support

jcampbell05, 2023-07-31:
Some applications will encode the RAW data as JPEG data to keep the file size small and to allow for faster decoding on the CPU when exporting as a DNG.

This required use of the level 2 opcodes otherwise these images will often come out as purple. It doesn't appear that rawspeed supports these and therefore an end user will need to use Adobe's DNG SDK to decode them.

LebedevRI, 2023-07-31:
By JPEG, do you literally mean `.JPEG`, or `.DNG` with `compression =  34892: Lossy JPEG`?
If latter, this is most certainly a duplicate of https://github.com/darktable-org/rawspeed/issues/215 & friends,
because in general, stage-2 DNG Opcodes are handled:
https://github.com/darktable-org/rawspeed/blob/9c5e0f97a1816d2b1538a1a508092b8dd7e53164/src/librawspeed/decoders/DngDecoder.cpp#L608-L622

Can you contribute a problematic sample to the https://raw.pixls.us/ please?

jcampbell05, 2023-07-31:
`.DNG` with JPEG lossy compression. This is the metadata from RawDigger, this file when opened in Darktable which uses latest raw speed ends up with the purple.

[image]

Unfortunately I can't share that specific file since I don't have permission however the issue appears to be that it's a "Lightroom fastload DNG". From what I understand raw.pixls.us isn't the correct place to submit a sample for that since you only want DNGs from the original camera and not any from processing software

LebedevRI, 2023-07-31:
> Unfortunately I can't share that specific file since I don't have permission however the issue appears to be that it's a "Lightroom fastload DNG". From what I understand raw.pixls.us isn't the [...]

Yes, i know what RPU says about such DNG's, but this is clearly an exception to that rule :)
I'm afraid someone would need to contribute such an affected sample ther,
otherwise this can not really be resolved.

jcampbell05, 2023-08-01:
Okay, let me see if I can submit something that re-creates the issue and we have permission to share.

LebedevRI, 2023-08-01:
Yes, please, thank you!

kmilos, 2023-08-02:
What about setting you black levels manually to 22 23 23?

jcampbell05, 2023-10-02:
> What about setting you black levels manually to 22 23 23?

Is there an easy way to do that without building the library, using Darktable ?