darktable-org/rawspeed #507: Support for the new Sony A6700 please :)

KevinDAdam, 2023-08-02:
Hi,

Can we please have support for the new Sony A6700 camera (ILCE-6700)?
I've uploaded 2 samples to https://raw.pixls.us/ - one lossless compressed and one lossy compressed.

Thanks,
Kevin.

LebedevRI, 2023-08-02:
Thank you for contributing the samples!

kmilos, 2023-08-02:
Please also contribute the uncompressed mode to complete the full set.

I'll verify the input colour matrix as soon as ADC adds it...

KevinDAdam, 2023-08-02:
> Please also contribute the uncompressed mode to complete the full set.
> 
> I'll verify the input colour matrix as soon as ADC adds it...

The A6700 only has 2 modes - lossless compressed and lossy compressed.

kishkaru, 2023-08-06:
Willing to provide additional samples if needed.

kmilos, 2023-08-06:
> I wonder why? Market segmentation with future model?

Something like that, looks like it's reserved for the A7 class...

zinger565, 2023-08-08:
Fellow a6700 owner. Is there anything needed that I could provide?

Thanks.

HStep20, 2023-08-09:
So, I am completely new at this, but also blocked on some work by no editor supporting the camera. I simply added this to my cameras.xml file, and it seems to be able to read them:

```
	<Camera make="SONY" model="ILCE-6700">
		<ID make="Sony" model="ILCE-6700">Sony ILCE-6700</ID>
		<CFA width="2" height="2">
			<Color x="0" y="0">RED</Color>
			<Color x="1" y="0">GREEN</Color>
			<Color x="0" y="1">GREEN</Color>
			<Color x="1" y="1">BLUE</Color>
		</CFA>
		<Crop x="0" y="0" width="0" height="0"/>
		<Sensor black="512" white="16383"/>
		<ColorMatrices>
			<ColorMatrix planes="3">
				<ColorMatrixRow plane="0">7657 -2847 -607</ColorMatrixRow>
				<ColorMatrixRow plane="1">-4083 11966 2389</ColorMatrixRow>
				<ColorMatrixRow plane="2">-684 1418 5844</ColorMatrixRow>
			</ColorMatrix>
		</ColorMatrices>
	</Camera>
```

However, If something were broken, I am not sure exactly what I should be looking for in order to see what isn't working properly with adding support via the cameras.xml file. Based on @kmilos comment, I expect that we need to properly verify the Color Matrix before it becomes official (this is just the 6600 color matrix), but it seems okay to use in the meantime for having workable files

kmilos, 2023-08-10:
> (this is just the 6600 color matrix), but it seems okay to use in the meantime for having workable files

These are almost certainly not the right values as the sensor is different.

KevinDAdam, 2023-08-10:
> So, I am completely new at this, but also blocked on some work by no editor supporting the camera. I simply added this to my cameras.xml file, and it seems to be able to read them:
>
> ```
> 	<Camera [...]

I'm using a different, but also working config:
------------------------
	<Camera make="SONY" model="ILCE-6700">
		<ID make="Sony" model="ILCE-6700">Sony ILCE-6700</ID>
		<CFA width="2" height="2">
			<Color x="0" y="0">RED</Color>
			<Color x="1" y="0">GREEN</Color>
			<Color x="0" y="1">GREEN</Color>
			<Color x="1" y="1">BLUE</Color>
		</CFA>
		<Crop x="0" y="0" width="-28" height="0"/>
		<Sensor black="512" white="16380"/>
		<ColorMatrices>
			<ColorMatrix planes="3">
				<ColorMatrixRow plane="0">6972 -2408 -600</ColorMatrixRow>
				<ColorMatrixRow plane="1">-4330 12101 2515</ColorMatrixRow>
				<ColorMatrixRow plane="2">-388 1277 5847</ColorMatrixRow>
			</ColorMatrix>
		</ColorMatrices>
	</Camera>
------------------------
Mine is taken from a copy of the FX30 config.

closed 2023-08-24 by LebedevRI