darktable-org/rawspeed #516: Support JPEG XL compression in DNG

kmilos, 2023-09-01:
Just added in DNG 1.7

cytrinox, 2023-09-01:
Good hint. Disappointed that they only specified it for LinearRaw but not for CFA.

kmilos, 2023-11-23:
For example, [appears as the enhanced linear raw subimage in new Adobe AI denoised DNGs](https://discuss.pixls.us/t/denoise-comparison-darktable-on1-and-topaz/40563/17?u=kmilos).

kmilos, 2024-01-19:
And now also looks like it's in [Samsung Galaxy enhanced ExpertRAW DNGs](https://r2.community.samsung.com/t5/CamCyclopedia/JPEG-XL-Image-Codec/ba-p/15356525).

golemusu, 2025-06-18:
+10.

I have already converted my whole RAW collection to DNG 1.7 lossy (with Adobe DNG converter) as it reduces space consumption by 50-60% (or even 80% if it is uncompressed DNG like in Galaxy S21fe). In couple of A/B tests I did with Adobe Lightroom I could not see a difference between lossless DNG and much smaller lossy DNG.

So now I have from each pic a lossless DNG (typically 10-12MB as it is 12mpx) and lossy DNG 1.7 (typically 4-5MB). I will delete all lossless versions and gain about 50GB of HD space but will have to wait until Darktable and couple of other programs have DNG 1.7 support.

If you are professional photographer with much larger collection you can probably save hundreds or even thousands of GB of space by converting "not so important" pics to DNG lossy! And I am already replying in advance for potential "storage is cheap comments", it is actually not that cheap if you have your stuff in cloud. And being able to send the files more than 50% faster is also very appealing.

There is also another very good feature with DNG 1.7, when you convert with Adobe DNG converter to DNG 1.7 lossy you can also reduce resolution. So if you have huge 50mpx or larger RAW pics you can reduce space consumption by up to 95% ! if you drop the resolution also to 10-12mpx which is probably good enough for 95% of applications.

I read from LibRaw changelog that DNG 1.7 support (with JXL comp) should already be in LibRaw so is there any plans of bringing this to DarkTable?

golemusu, 2025-06-18:
Quote from different topic but discussion was requested to move here:

> No, because LibRaw would have to be built w/ Adobe DNG SDK:
> 
> "DNG 1.7 (including JPEG-XL compression) support via Adobe DNG SDK 1.7.x integration"

@kmilos Can you elaborate what do you mean by this?

I was under the impression that everything related to JXL is open source / royalty free.

cytrinox, 2025-07-11:
I've done some experiments, it seems pretty easy to implement jpeg-xl support into DNGLab. I've generated a set of JPEG-XL encoded DNGs with Adobe DNG Converter and was able to convert them with dnglab to native LJPEG compression DNGs.
The problematic part is that with lossy JPEG-XL DNG files from Adobe, Opcode2 processing is required.

For lossy JPEG-XL compressed DNGs (both mosaic or linear) a PolynominalMap (id=8) opcode is embedded in OpcodeList2. Without processing this opcode, the colors are garbage.

PolynominalMap is only used for lossy files, this opcode does not appear in lossless files yet.

I leave this as hint here 😄

cytrinox, 2025-07-12:
The JPEG-XL DNG sample set is available here: https://rawdb.dnglab.org/dng/compression-sets/

Forza-tng, 2025-10-05:
Hi, I uploaded to https://raw.pixls.us/ a 50MP and a 12MP DNG  created using ExpertRAW (JPEG-XL DNG) on a Samsung S24 Ultra. This version is, I think different than S23 and earlier - at least I cannot open them in Darktable or other tools like ImageMagick as I can with DNG from earlier phones.