darktable-org/rawspeed #562: WXS schema failed to compile

aurelienpierre, 2023-11-18:
On Windows (Github image windows-latest ) with:

- GNU GCC 13.2.0,
- LibXml2: 2.12.0
- Pugixml 1.14

```
FAILED: lib/ansel/rawspeed/data/cameras.xml.touch D:/a/ansel/ansel/build/lib/ansel/rawspeed/data/cameras.xml.touch 
cmd.exe /C "cd /D D:\a\ansel\ansel\build\lib\ansel\rawspeed\data && D:\a\_temp\msys64\ucrt64\bin\xmllint.exe --nonet --valid --noout --schema D:/a/ansel/ansel/src/src/external/rawspeed/data/cameras.xsd D:/a/ansel/ansel/src/src/external/rawspeed/data/cameras.xml && D:\a\_temp\msys64\ucrt64\bin\cmake.exe -E touch D:/a/ansel/ansel/build/lib/ansel/rawspeed/data/cameras.xml.touch"
D:/a/ansel/ansel/src/src/external/rawspeed/data/cameras.xsd:195: element complexType: Schemas parser error : complex type 'CFA2Type': The content model is not determinist.

WXS schema D:/a/ansel/ansel/src/src/external/rawspeed/data/cameras.xsd failed to compile
```

The previous successful run was with LibXml2 2.11.5.

LebedevRI, 2023-11-18:
In the mean time, there's `-DUSE_XMLLINT=OFF` escape hatch.

closed 2023-11-18 by LebedevRI

LebedevRI, 2023-11-18:
@aurelienpierre thank you.
Fixed in `develop`, and propagated to `stable` (which is the recommended branch to use FYI).

PR #563 (2023-11-18): Attempt to fix XSD validation w/ libxml2 2.12.0
Validation for `CFA2` variant of CFA encoding is rather partial anyways,
but it isn't even legal to try to match the same element as different XSD types...

Fixes #562.