darktable-org/rawspeed #695: Support for Canon Powershot S100

arturoisilvia, 2024-03-18:
I request support for Canon Powershot S100 with the following parameters:
Level 0 -> 128
Level 1 -> 128
Level 2 -> 128
Level 3 -> 128
white point -> 2990

Thank you.

gi-man, 2024-03-18:
For more context:  https://discuss.pixls.us/t/purple-image-from-canon-powershot-s100-cr2/42640/14

It seems like reading the white/black values from CR2 for this camera is incorrect since 4.6.1. Maybe an effect from https://github.com/darktable-org/rawspeed/pull/578

kmilos, 2024-03-18:
Just to clarify, this is a request to revert part of https://github.com/darktable-org/rawspeed/pull/578 for S100 only.

LebedevRI, 2024-03-18:
That change is correct, we read reasonable values from the makernotes:
```
whitePoint: 2976
blackLevelSeparate: (2 x 2) 128 128 128 128
```
what isn't however correct, is the rest of the hardcoded stuff,
because after sampling the hardcoded black areas we get:
```
blackLevelSeparate: (2 x 2) 65535 65535 65535 65535
```
Refs. #389.