darktable-org/rawspeed #712: question: Difference for embedded lens-correction for OM-System Mark II between DNG and ORF

astmuc, 2024-04-19:
I can see the automatic lens correction using embedded info if the input is DNG. But i see no lens correction  for the native ORF format. 

Is this an effect of https://github.com/darktable-org/rawspeed/issues/686, or is it maybe simply an unsupported feature ?

kmilos, 2024-04-19:
Lens correction is out of scope for the RawSpeed library.

Please seek support at https://discuss.pixls.us/c/software/darktable/19 or https://github.com/darktable-org/darktable/issues

kmilos, 2024-04-19:
> Is this an effect of...

It is not.

closed 2024-04-19 by kmilos

LebedevRI, 2024-04-19:
None of the lens correction opcodes are supported by rawspeed:
https://github.com/darktable-org/rawspeed/blob/bd2b74eb46446924a8f5b15f58fdad18a6f6e3d3/src/librawspeed/common/DngOpcodes.cpp#L748-L754
https://github.com/darktable-org/rawspeed/blob/bd2b74eb46446924a8f5b15f58fdad18a6f6e3d3/src/librawspeed/common/DngOpcodes.cpp#L771-L772
so @kmilos is correct.