darktable-org/rawspeed #819: OM Digital Solutions OM-3 is missing RPU sample coverage

LebedevRI, 2025-04-27:


sarunasb, 2025-04-27:
A set of samples just uploaded.

LebedevRI, 2025-04-27:
> A set of samples just uploaded.

Thank you! I've verified the set, but it's a bit unfortunate that they are handheld.

closed 2025-04-27 by LebedevRI

sarunasb, 2025-04-27:
Right, OM-3 offers all three hi-res modes in both handheld and tripod modes. However in tripod mode ORF is always made at 80Mpx, regardless of Mpx selection, only the OOC JPGs are rendered at 25, 50 or 80Mpx as set. I have just uploaded 12 and 14 bit samples to RPU.

LebedevRI, 2025-04-27:
I was actually talking about the fact that those shots were taken when the camera was not fixed to a stable surface.
Verified the new uploads too, thank you!

astmuc, 2025-05-08:
I have a question about OM-3  support for the next release. The current version of cameras.xml in develop branch marks the OM-3 as in being in danger zone for missing RPU files. But this ticket is closed. Is something missing to ensure that the camera is supported in the next release ?

LebedevRI, 2025-05-08:
> The current version of cameras.xml in develop branch marks the OM-3 as in being in danger zone for missing RPU files.

I'm waiting on https://github.com/darktable-org/rawspeed/issues/820
before doing a pass and marking cameras with samples as such,
and updating stable branch.