darktable-org/rawspeed #882: API request: Interface for stealing the data buffer out of a RawImage.

mcourteaux, 2025-08-02:
When loading files that do not contain a mosaiced image, but are already full-color, it'd be convenient if we can steal the data allocation (sitting in the `std::vector<uint8, ...> data`) out of the `RawImage` object. That would prevent a pointless memcpy.

The only problem I see with this right now is that the `std::vector` uses a custom allocator, so stealing the resource would require us to be careful to also free it with that same allocator.

Thoughts on this @LebedevRI?

LebedevRI, 2025-08-02:
> When loading files that do not contain a mosaiced image, but are already full-color, it'd be convenient if we can steal the data allocation (sitting in the `std::vector<uint8, ...> data`) out of the `RawImage` object. That would prevent a pointless memcpy.

It would look something like
```
std::vector<...> RawImage::takeUnderlyingBuffer() {
  std::vector<...> vec;
  std::swap(this->buf, vec);
  return vec;
}
```

> The only problem I see with this right now is that the `std::vector` uses a custom allocator, so stealing the resource would require us to be careful to also free it with that same allocator.

I'm not following. It's currently not `std::unique_ptr`, you can't construct it from a pointer, nor can you take the underlying owned memory pointer from it without it being deallocated by `vector`.
 
> Thoughts on this [@LebedevRI](https://github.com/LebedevRI)?

I'm moving forward with Rust port. I'm not particularly interested in non-`cameras.xml`-only changes now.

mcourteaux, 2025-08-02:
Okay thanks for the quick reply. How is the Rust port going? Is it usable? Any particular functional differences you are working on?

closed 2025-08-02 by mcourteaux

LebedevRI, 2025-08-02:
> Okay thanks for the quick reply. 

> How is the Rust port going?

As good as one could expect, for a completely new language for me.

> Is it usable?

It is not yet usable at this time.

> Any particular functional differences you are working on?

I'm not really sure what you mean.
The idea would be to port things over in small self-contained chunks
that result in more RPU samples being decoded identically.